MIT-SPARK/TEASER-plusplus #79: Performance worse than RANSAC

saijadhav9496, 2021-02-23:
**Have you read the documentation?**

- [x] Yes
- [] No - then this issue will be closed.

I am implementing TEASER in python for pose estimation. I provided 1000 correspondences generated by FCGF features for two similar point clouds with approximately 5000 points with random rotation and translation added. TEASER's performance is consistently worse than RANSAC for the same correspondences. I have kept the hyper parameters at their default values. I am running with OMP_NUM_THREADS=12 otherwise the code runs into a segmentaion fault. The GNC-TLS solver runs for approximately 12 iterations for rotation estimation before converging.

The rotation error I get from TEASER is ~2(radians) and RANSAC gives ~0.02. I am wondering if I am doing something wrong here.

jingnanshi, 2021-02-23:
Hi @saijadhav9496 thank you for your interest. You need to provide more information for me to help you. Please provide some code snippet showing how you perform the tests, and take a look at this as well https://github.com/MIT-SPARK/TEASER-plusplus/tree/master/examples/teaser_python_fpfh_icp.

jingnanshi, 2021-03-03:
Closing due to no further replies.

closed 2021-03-03 by jingnanshi