MIT-SPARK/TEASER-plusplus #80: CMake Error at teaser/CMakeLists.txt:21 (add_library):

tsigit, 2021-02-24:
When running the command
`cmake .. `
from C:\Program Files\TEASER-plusplus\build on Windows10 (Python3.6.2-64bit, Visual Studio 15 2017 Win64)

all the other builds (googletest, pmc, spectra, tinyply) run without error, except for the pybind11, which throws lot of CMake Error like this one

```
CMake Error at teaser/CMakeLists.txt:21 (add_library):
  Target "teaser_registration" links to target "Eigen3::Eigen" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
```
and all subsequent commands for installing python bindings of Teaser++ on Windows fails

jingnanshi, 2021-02-25:
Hi @tsigit , thank you for your interest. The error looks like something related to the Eigen library. I'm not familiar with Windows 10 so you probably need to figure this out yourself.

tsigit, 2021-02-25:
Thanks @jingnanshi 
I followed some of the steps from [here](https://github.com/zhongjingjogy/use-eigen-with-cmake) and have been able to get through the Cmake and Eigen errors

closed 2021-03-03 by jingnanshi