MIT-SPARK/TEASER-plusplus #85: [QUESTION] Angle error computation?

adrian11655, 2021-03-05:
Hello, 
The following is the angle error calculation function in the code. Is this method of calculating angle error reasonable? 
Three true (expected) Euler angles (true_theta1, true_theta2, true_theta3), are known, and three estimated Euler angles (estimated_theta1, estimated_theta2, estimated_theta3) can be computed through the estimated Rotation Matrix. Why is the total angle error not calculated by the error of three Euler angles? That is, 
Angle_error = sqrt[(estimated_theta1-true_theta1)^2+(estimated_theta2-true_theta2)^2+(estimated_theta3-true_theta3)^2]

def get_angular_error(R_exp, R_est):
    """
    Calculate angular error
    """
    return abs(np.arccos(min(max(((np.matmul(R_exp.T, R_est)).trace() - 1) / 2, -1.0), 1.0)));

jingnanshi, 2021-03-05:
@adrian11655 Thank you for your interest. What you said, while is a rough estimate, is typically not used, because there are different definitions of Euler angles, and it is not clear which definition you are using. The angular error we are using follows directly from the definition of geodesic distance in SO(3). Geometrically, it is equivalent to the angle of rotation around some axis in 3D. See this paper for more info: https://www.cs.cmu.edu/~cga/dynopt/readings/Rmetric.pdf

adrian11655, 2021-03-06:
Thank you very much.

closed 2021-03-06 by adrian11655