MIT-SPARK/TEASER-plusplus #86: [QUESTION] Finding the transform of an object in a point cloud scene

saikishor, 2021-03-07:
**Have you read the documentation?**

- [x] Yes
- [ ] No - then this issue will be closed.

Hello,

First of all, thank you for such a great work. It is pretty impressive. I am trying to understand how to use TEASER with my use case. I would like to get the transform to a particular object in a scene. I have a .pcd info of my object of interest and I have a wholesome pointcloud with not only the object but with more information about it's surroundings too. What I am interested here is the transform to the object. 

I do see that if I give some information of different columns in the solve method. It blows up very badly, and then I have seen some information in other issues, that you were explaining them that we need to input the correspondences to the `solve` method in this case. I am interested to know how to compute these correspondences, because in my case, I might not have the whole pointcloud information of the object from the scene, so I am not sure how to exactly input the corresponding information in both `src` and `dst` to compute the transform properly. 

Could you give me some insights on how to proceed?. Thanks

jingnanshi, 2021-03-08:
Hi @saikishor , one way to do what you said is to use an object detector to first locate your object of interest in the 3D scene. Then, use the segmented points with a feature extraction algorithm to calculate the feature points, and then match the calculated features with those of the object model you have.

saikishor, 2021-03-08:
@jingnanshi Thanks for a quick response.
The thing is I already have an isolated cloud of the object from the object detection as you mentioned, but how can I extract these features. I am not sure how to extract the proper features from my cloud and the template cloud I have. 

Could you explain me how I can do that with the TEASER plusplus?

jingnanshi, 2021-03-08:
@saikishor TEASER++ has bindings for using PCL FPFH features. In addition, there are other libraries that you can use (for example https://github.com/PointCloudLibrary/pcl and www.open3d.org/) to get features.

saikishor, 2021-03-08:
OK i will give it a shot. Thank you.