MIT-SPARK/TEASER-plusplus #92: [QUESTION] test on two ply but the result is not good

david-ngo-sixense, 2021-04-14:
**Have you read the documentation?**

- [ ] Yes
- [x] No - then this issue will be closed.

Thanks for your code and paper. It is very interesting.
I installed teaser++ on windows (with some modifications). I got the two examples (*.cc) running. 
I tested on two point cloud with teaser_cpp_fpfh but the result was not good (the point cloud is so simple and symmetric).
I think the performance depends on keypoint detection, descriptor extraction.
teaser_cpp_ply takes a lot of time because it takes all points in the cloud as keypoint. [

](url)
Can you please test your algorithms on these clouds ? 
Thanks.

jingnanshi, 2021-04-16:
Hi @david-ngo-sixense , you need to first generate correspondences & then feed the correspondences to TEASER++. I recommend you to use the Python interface together with Open3D so that you can test different feature detectors and select the best combination.

david-ngo-sixense, 2021-04-17:
Hi @jingnanshi , thanks for your reply. 
I installed your teaser++ (c++ version, on a windows machine) and used it sucessfully for a C++ project.
Yes, I want to use the python interface, but I can't install the python wrapper.

`(Pytorch) C:\dev\TEASER-plusplus\build\python>pip install .
Processing c:\dev\teaser-plusplus\build\python
Building wheels for collected packages: teaserpp-python
  Building wheel for teaserpp-python (setup.py) ... done
  Created wheel for teaserpp-python: filename=teaserpp_python-1.0.0-py3-none-any.whl size=18144 sha256=631c0432c1ba85f575c7cace87144edbc03ab4154435b71693b54234290699cf
  Stored in directory: C:\Users\ttngo\AppData\Local\Temp\pip-ephem-wheel-cache-f485dfpb\wheels\d7\cc\2e\d79e3ca0e0949697fc37003fde9b332b17db6e371bdd357de5
Successfully built teaserpp-python
Installing collected packages: teaserpp-python
Successfully installed teaserpp-python-1.0.0

(Pytorch) C:\dev\TEASER-plusplus\build\python>python
Python 3.8.8 (default, Feb 24 2021, 15:54:32) [MSC v.1928 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import teaserpp_python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\dev\TEASER-plusplus\build\python\teaserpp_python\__init__.py", line 1, in <module>
    from .teaserpp_python import *
ModuleNotFoundError: No module named 'teaserpp_python.teaserpp_python'
>>>`
Do you know why ?

jingnanshi, 2021-04-17:
Make sure when you compile the C++ project you turn on BUILD_PYTHON_BINDINGS

closed 2021-07-17 by jingnanshi

manavkulshrestha, 2023-02-27:
Hi, I don't have too much experience with C++. How would I go about turning on BUILD_PYTHON_BINDINGS when compiling the project? I just followed the instructions for installation over at https://teaser.readthedocs.io/en/master/installation.html and it did not mention anything about the flag you mention