MIT-SPARK/TEASER-plusplus #93: Use PCL::Visualization with Teaser result

silviadonzelli, 2021-04-16:
Hi,
I want to see my result using pcl::visualization but I'm a little bit confused. With SAC-IA, I can visualize sac_ia as the result of sac_ia.align(*cloudSac), but with Teaser++ what can I use?
Thank you in advance.

jingnanshi, 2021-04-16:
Hi @silviadonzelli , we don't provide visualization capability in C++. You need to either write your own code to visualize using PCL:visualization, or use the Python interface and visualize the points using Open3D.

silviadonzelli, 2021-04-17:
Hi, thank you. So, with my Rotation and Translation I could do the Rototranslation matrix and then? I have the code to visualize point cloud but I don't know how use Teaser result.
How can I do?
Thank you very much,

closed 2021-05-31 by silviadonzelli