MIT-SPARK/TEASER-plusplus #95: [QUESTION]libpmc.so: cannot open shared object file: No such file or directory

s20163081161, 2021-04-23:
When I run ./teaser_cpp_ply, i got "error while loading shared libraries: libpmc.so: cannot open shared object file: No such file or directory"

But I can find libpmc.so in the directory "TEASER-plusplus/build/pmc-build".

Whta's wrong ?

jingnanshi, 2021-04-23:
Try run `sudo ldconfig`

s20163081161, 2021-04-23:
include(ExternalProject)
ExternalProject_Add(**pmc**
        GIT_REPOSITORY    https://github.com/pybind/**pybind11**.git

In the file "pybind11.CMakeLists.txt.in"，the name seems to be wrong ?

jingnanshi, 2021-04-23:
Good catch! You can try changing `pmc` to `pybind11` and see whether it solves this issue.

s20163081161, 2021-04-27:
@jingnanshi 
Using a pointcloud whose size is 16328, I got stuck in the step "max_clique_ = clique_solver.findMaxClique(inlier_graph_);" and finally was killed.

The algorithm do not allow big pointclouds ???

jingnanshi, 2021-04-29:
@s20163081161 TEASAER++ takes in correspondences, not raw points. You need to use a feature detector and matcher to find correspondences.

jingnanshi, 2021-07-17:
Doc fixed in PR #107