MIT-SPARK/TEASER-plusplus #96: Ctest (Test #18) failed

kidpaul94, 2021-05-06:
Hi, thank you for the great work. 
I'm merely following the documentation about compilation and installation, and encountered this error:
    
    98% tests passed, 1 tests failed out of 48

    Total Test time (real) =  32.66 sec

    The following tests FAILED:
	     18 - RegistrationTest.SolveRegistrationProblemDecoupled (SEGFAULT)
    Errors while running CTest

I'm not sure where this error comes from. Does this test use any PCL function? (I didn't download PCL)

Current system version:
Ubuntu 20.04
CMake = 3.16.3
Eigen3 = 3.3.7
Boost = 1.71.0

kidpaul94, 2021-05-06:
OMP_NUM_THREADS=12 was still the only solution for this.

closed 2021-05-06 by kidpaul94

jingnanshi, 2021-05-08:
@kidpaul94 thank you for reporting the issue. You are correct, unfortunately using 12 threads is still the only solution.

SubMishMar, 2022-03-22:
Where can I set `OMP_NUM_THREADS=12`, for me 20 tests fail.
[image]
`