MIT-SPARK/TEASER-plusplus #100: [BUG] build teaser++ and run ctest == error #18

knoname, 2021-05-26:
**Describe the bug**
I am attempting to run ctest after building.  All tests pass on two different machines (iMac and Ubuntu box).  Both fail the same test #18.  I'm looking for suggestions what I might be doing wrong??

Thank you for any insights you can offer!

cmake and make completed with no errors.

**Have you run the unit tests?**
yes.

**Installed Dependencies**
Ubuntu 20.04.2 LTS
64G memory
Intel® Xeon(R) Silver 4114 CPU @ 2.20GHz × 40 
cmake version 3.16.3
eigen 3.3.7
boost 1.71
pcl 1.10.0

**To Reproduce**
sudo apt install cmake libeigen3-dev libboost-all-dev libpcl-dev
git clone https://github.com/MIT-SPARK/TEASER-plusplus.git
cd TEASER-plusplus && mkdir build
cd build
cmake ..
make
OMP_NUM_THREADS=12
ctest

**Additional context**
detailed console output from build, ctest and ctest -VV

[teaser-plusplus_ctest_error.txt](https://github.com/MIT-SPARK/TEASER-plusplus/files/6549942/teaser-plusplus_ctest_error.txt)

jingnanshi, 2021-06-29:
Hi @knoname thanks for reporting the bug. I've seen the same issue before unfortunately I could not isolate the cause within reasonable time. Feel free to take a look yourself and I welcome any PR.

jingnanshi, 2023-05-22:
This should have been fixed with #165 and the fixes in my fork of the PMC library.

closed 2023-05-22 by jingnanshi