MIT-SPARK/TEASER-plusplus #99: [QUESTION]Trying Teaser++ with two .ply files

tkzky, 2021-05-26:
The code is very amazing!
But when I try Teaser++ with two .ply files, it aborts during the method solve.
Can provide some some tutorials about how to use 3DSmoothNet to generate correspondences?
Thank you!

jingnanshi, 2021-05-26:
@tkzky take a look at this file: https://github.com/MIT-SPARK/TEASER-plusplus/blob/master/examples/teaser_python_3dsmooth/teaser_python_3dsmooth.py

tkzky, 2021-05-27:
Thanks for your reply!
But when i reproduce the GIF,  after "python teaser_python_3dsmooth.py", there is an error:
"==================================================
        TEASER++ Python registration example      
==================================================
TEASER++ Parameters are: <Params with noise_bound=0.05
cbar2=1
estimate_sccaling=0
rotation_estimation_algorithm=GNC-TLS
inlier_selection_mode=PMC_EXACT
>
Starting scale solver.
Scale estimation complete.
Max core number: 76
Num vertices: 841
*** [pmc heuristic: thread 5]   current max clique = 62,  time = 8.4877e-05 sec
*** [pmc heuristic: thread 6]   current max clique = 66,  time = 0.000148058 sec
[pmc heuristic]	 mc = 66
Created adjacency matrix in 0.000236034 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 840, |E| = 39368
[pmc: initial k-core pruning]  after pruning:  |V| = 840, |E| = 4137
[pmc]  initial pruning took 0.000172138 sec
[pmc: sorting neighbors]  |E| = 8274, |E_sorted| = 8274
|V| = 94
-----------------------------------------------------------------------
Max Clique of scale estimation inliers: 
Using chain graph for GNC rotation.
Starting rotation solver.
GNC rotation estimation noise bound:0.1
GNC rotation estimation noise bound squared:0.01
GNC-TLS solver terminated due to cost convergence.
Cost diff: 0
Iterations: 17
Rotation estimation complete.
Starting translation solver.
Translation estimation complete.
Max clique size: 66
GT Inliers count: 34
GT Inliers: {130, 264, 265, 18, 275, 530, 151, 792, 281, 409, 291, 422, 811, 178, 565, 319, 195, 579, 583, 332, 343, 599, 601, 222, 481, 612, 104, 107, 108, 497, 114, 754, 249, 637}
Now drawing correspondences ...
terminate called after throwing an instance of 'pybind11::error_already_set'
  what():  SystemError: <built-in method __repr__ of PyCapsule object at 0x7fadd34d4090> returned a result with an error set
"
How can i solve it? Looking forward to your reply! Thank you!

russellyq, 2021-06-29:
Hi,

I got the same error. Did you fix it ?

Thanks.

jingnanshi, 2021-06-29:
@russellyq @tkzky can you report the version of pybind11 you have installed? Thanks

russellyq, 2021-06-30:
> h

It is the same version in your packages. V2.4.3

HOUYONGKUO, 2021-08-02:
Hi,
I got the same error. Did you fix it ?
The version of pybind11:2.6.1 
Thanks.

ylllllll, 2022-11-28:
there is a ";" in examples/teaser_python_3dsmooth/teaser_python_3dsmooth.py 112， delete it.