MIT-SPARK/TEASER-plusplus #102: The teaser_python_3dsmooth.py issue

yukailin-git, 2021-07-06:
==================================================
        TEASER++ Python registration example      
==================================================
TEASER++ Parameters are: <Params with noise_bound=0.05
cbar2=1
estimate_sccaling=0
rotation_estimation_algorithm=GNC-TLS
inlier_selection_mode=PMC_EXACT
>
Starting scale solver.
Scale estimation complete.
Max core number: 76
Num vertices: 841
*** [pmc heuristic: thread 2]   current max clique = 66,  time = 9.89437e-05 sec
[pmc heuristic]  mc = 66
Created adjacency matrix in 0.000591993 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 840, |E| = 39368
[pmc: initial k-core pruning]  after pruning:  |V| = 840, |E| = 4137
[pmc]  initial pruning took 0.00275898 sec
[pmc: sorting neighbors]  |E| = 8274, |E_sorted| = 8274
|V| = 94
-----------------------------------------------------------------------
Max Clique of scale estimation inliers: 
Using chain graph for GNC rotation.
Starting rotation solver.
GNC rotation estimation noise bound:0.1
GNC rotation estimation noise bound squared:0.01
GNC-TLS solver terminated due to cost convergence.
Cost diff: 0
Iterations: 17
Rotation estimation complete.
Starting translation solver.
Translation estimation complete.
Max clique size: 66
GT Inliers count: 34
GT Inliers: {130, 264, 265, 18, 275, 530, 151, 792, 281, 409, 291, 422, 811, 178, 565, 319, 195, 579, 583, 332, 343, 599, 601, 222, 481, 612, 104, 107, 108, 497, 114, 754, 249, 637}
Now drawing correspondences ...
terminate called after throwing an instance of 'pybind11::error_already_set'
  what():  SystemError: <built-in method __repr__ of PyCapsule object at 0x7f16dae14ba0> returned a result with an error set

Hi, author
When I run the file, I got an error like this.
Do you know what is the problem?
Best Regards, 
Yu-Kai Lin

jingnanshi, 2021-07-06:
related #99

jingnanshi, 2021-07-06:
HI @yukailin-git thanks for your interest. I will take a look soon.

fyang064, 2021-07-29:
I'm curious about your ground truth file. How did you get the gt.log file after running the 3DSmooth generator? Looking forward to hearing from you, thank you!

jingnanshi, 2021-07-29:
@fyang064 gt.log comes from the 3DMatch dataset itself.

HOUYONGKUO, 2021-08-03:
> 嗨@yukailin-git感谢您的关注。我很快就会去看看。

Hi！
I also encountered the same problem, have you found the cause of this problem?
Thanks

Pale-cream, 2022-02-02:
此问题现在依然无法解决

closed 2022-02-04 by yukailin-git

ljjTYJR, 2022-03-08:
One possibility is that the _open3d_ is not compatible.
Try:
`conda uninstall open3d`  # uninstall old open3d
and
`conda install -c open3d-admin open3d=0.10.0`

Then try again.

G-rookie, 2022-05-19:
@yukailin-git Hi！
I also encountered the same problem, have you found the cause of this problem and do you know how to solve it ?
I hope you can give me a hand ,Thank you !