MIT-SPARK/TEASER-plusplus #103: Target "teaser_cpp_fpfh" links to target "teaserpp::teaser_features" but   the target was not found[BUG]

SharineLee, 2021-07-09:
TEASER-plusplus/examples/teaser_cpp_fpfh/build$ cmake ..

-- Configuring done
CMake Error at CMakeLists.txt:14 (add_executable):
  Target "teaser_cpp_fpfh" links to target "teaserpp::teaser_features" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

-- Generating done
-- Build files have been written to: /home/sharine/GroundTextureEVOProject/TEASER-plusplus/examples/teaser_cpp_fpfh/build

This is my CMakeLists.txt:
`cmake_minimum_required(VERSION 3.10)
project(teaser_cpp_fpfh)

set (CMAKE_CXX_STANDARD 14)

set(PCL_DIR "/usr/local/PCL19/share/pcl-1.9")

find_package(Eigen3 REQUIRED)
find_package(teaserpp REQUIRED)
find_package(Boost 1.58 REQUIRED)
find_package(PCL 1.9 REQUIRED COMPONENTS common features kdtree)

# Change this line to include your own executable file
add_executable(teaser_cpp_fpfh teaser_cpp_fpfh.cc)

# Link to teaserpp & Eigen3
target_link_libraries(teaser_cpp_fpfh 
  Eigen3::Eigen 
  teaserpp::teaser_registration 
  teaserpp::teaser_features
  teaserpp::teaser_io 
  ${PCL_LIBRARIES})

# Copy the data files to build directory
file(COPY ../example_data/
        DESTINATION ./example_data/
        FILES_MATCHING
        PATTERN *.ply)
`

b15h0y, 2021-07-09:
dealing with the same issue, any ideas how to solve it?

b15h0y, 2021-07-09:
solved it , you have to rebuild the library with `BUILD_TEASER_FPFH=ON` . it's turned `OFF` by default [Here](https://teaser.readthedocs.io/en/master/installation.html#available-cmake-options).

jingnanshi, 2021-07-10:
@b15h0y yes that's correct. Thanks for your reply.

closed 2021-07-10 by jingnanshi