MIT-SPARK/TEASER-plusplus #104: [QUESTION] Downsampling big clouds to find correspondences?

b15h0y, 2021-07-12:
**Have you read the documentation?**

- [x] Yes
- [ ] No - then this issue will be closed.

Is there a way to down-sample a point cloud with teaser++? I was calculating the correspondences between clouds and it takes a lot of time (3 h)! 

```
[INFO] Reading Point Clouds ..
 Read 436186 total vertices 
 Read 409625 total vertices 
 Read 418356 total vertices 
 Read 423251 total vertices 
 Finding correlations between : 0 , and 1
Calculating fpfh 
Calculating correspondences <- Stuck here
```

jingnanshi, 2021-07-12:
@b15h0y The FPFH correspondence computation module included in the TEASER++ repo is very rudimentary, and it does not have downsampling functions. For downsampling, I recommend either use PCL in C++, or Open3D in python.

b15h0y, 2021-07-12:
It would be really helpful to extend the Utils function in TEASER++ with methods to do conversions of point clouds to other libraries classes like `pcl::PointCloud<TYPE>::Ptr` or even an Eigen matrix to handle transformations better. So in a scenario like the one I am facing or in real application where the point cloud is huge some one can do something like this : 
```  
pcl::PCLPointCloud2::Ptr cloud = teaser::utils::toPCLPointCloud(cloud_read_with_teaserpp); <- Here
pcl::PCLPointCloud2::Ptr cloud_filtered (new pcl::PCLPointCloud2 ());
pcl::VoxelGrid<pcl::PCLPointCloud2> sor;
sor.setInputCloud (cloud);
sor.setLeafSize (0.01f, 0.01f, 0.01f);
sor.filter (*cloud_filtered);
teaser::PointCloud cloud_downsamples = teaser::utils::toTeaserppPointCloud(cloud_filtered); <- Back Here
...
```

jingnanshi, 2021-07-12:
I agree. I think this should be easy to do. If you submit a PR, I am more than happy to merge it.