MIT-SPARK/TEASER-plusplus #106: [BUG] Input data type for Matalb

jbalado, 2021-07-14:
I tried to run the example code in matlab with with src = rand(3,5). Same for dst.

src = rand(3,5);;
dst = rand(3,5);

cbar2 = 1;
noise_bound = 0.01;
estimate_scaling = false; % we know there's no scale difference
rot_alg = 0; % use GNC-TLS, set to 1 for FGR
rot_gnc_factor = 1.4;
rot_max_iters = 100;
rot_cost_threshold = 1e-12;

% The MEX function version
[s, R, t, time_taken] = teaser_solve_mex(src, dst, cbar2, ...
        noise_bound, estimate_scaling, rot_alg, rot_gnc_factor,    ...
        rot_max_iters, rot_cost_threshold);

% The MATLAB wrapper version
[s, R, t, time_taken] = teaser_solve(src, dst, 'Cbar2', cbar2,    'NoiseBound', noise_bound, ...
                                     'EstimateScaling',    estimate_scaling, 'RotationEstimationAlgorithm', rot_alg, ...
                                  'RotationGNCFactor', rot_gnc_factor, 'RotationMaxIterations', 100, ...
                                  'RotationCostThreshold', rot_cost_threshold);

The following error appeared: 

Undefined function 'teaser_solve' for input arguments of type 'double'.

I also had the same problem for 5x3 matrices, type single and PointCloud; and with teaser_solve_mex function.
Could you help me?

WJinhai, 2021-07-22:
I want to know How did you compile successfully in Matlab？

jingnanshi, 2021-07-22:
Hi @WJinhai @jbalado thanks for your interest. I will take a look at the MATLAB binding.

jbalado, 2021-07-22:
> I want to know How did you compile successfully in Matlab？

Sorry, I didn't say it before, but it's running on linux, a clean install with just the mapping toolbox (and the necessary ones).