MIT-SPARK/TEASER-plusplus #109: [QUESTION]

HOUYONGKUO, 2021-07-21:
Thanks for your work. I meet some issues using teaser++.
How do I get the keypoints data? If I want to register my own two data, but I don’t have keypoints, what should I do?
I would be grateful if you could help me figure this out.

jingnanshi, 2021-07-21:
Hi @HOUYONGKUO , thank you for your interest. You need to use a feature extractor & matcher to find correspondences. For example, you can use FPFH, or FCGF, or 3DSmoothNet.

HOUYONGKUO, 2021-08-12:
If there is no keypoints data, can Teaser++ be used for registration?

jingnanshi, 2021-08-12:
@HOUYONGKUO as described in the paper, you can try all-to-all with your input points, but that's only going to work for small point clouds.

LimHyungTae, 2024-04-03:
Voxelized-FPFH works well even though we do not extract any keypoints. Please follow the instructions in the papers for 3D sparse point clouds, not the Bunny dataset:
- https://arxiv.org/abs/2203.06612
- https://arxiv.org/abs/2311.00928

LimHyungTae, 2024-05-02:
If there are no further questions, we will close the issue :) Thx

closed 2024-05-02 by jingnanshi