MIT-SPARK/TEASER-plusplus #111: [BUG] Solver killed in teaser++ python

jbalado, 2021-07-26:
I ran the code for Minimal Python 3 example for linux and the result was as expected.

I modified the code to read two point clouds (25k points each with partial overlap) and perform the registration. The code reads and saves the parameters, but when I run the solver.solve it crashes and ends up being killed.

How can I fix this? 

```
# Libraries
import copy
import time
import numpy as np
import open3d as o3d
import teaserpp_python

NOISE_BOUND = 0.05

if __name__ == "__main__":
    print("==================================================")
    print("        TEASER++ Python registration example      ")
    print("==================================================")

    # Load data
    src_cloud = o3d.io.read_point_cloud("../example_data/Car_000000.txt",format='xyz')
    src = np.transpose(np.asarray(src_cloud.points))
    N = src.shape[1]
    
    dst_cloud = o3d.io.read_point_cloud("../example_data/s1.txt",format='xyz')
    dst = np.transpose(np.asarray(dst_cloud.points))

    # Populating the parameters
    solver_params = teaserpp_python.RobustRegistrationSolver.Params()
    solver_params.cbar2 = 1
    solver_params.noise_bound = NOISE_BOUND
    solver_params.estimate_scaling = False
    solver_params.rotation_estimation_algorithm = teaserpp_python.RobustRegistrationSolver.ROTATION_ESTIMATION_ALGORITHM.GNC_TLS
    solver_params.rotation_gnc_factor = 1.4
    solver_params.rotation_max_iterations = 100
    solver_params.rotation_cost_threshold = 1e-12

    solver = teaserpp_python.RobustRegistrationSolver(solver_params)
    start = time.time()
    solver.solve(src, dst)
    end = time.time()

    solution = solver.getSolution()

    print("=====================================")
    print("          TEASER++ Results           ")
    print("=====================================")

    print("Estimated rotation: ")
    print(solution.rotation)

    print("Estimated translation: ")
    print(solution.translation)

    print("Time taken (s): ", end - start)

```

jingnanshi, 2021-07-26:
@jbalado Thanks for your interest. It's probably crashing due to memory insufficiency. Note that the `src` and `dst` should be viewed as correspondences. In other words, a better way to do what you want to do is to extract corresponding points, then pass those points as Eigen matrices to TEASER++.

jbalado, 2021-07-27:
@jingnanshi I tried to applied the same code to both downsampled point clouds (4k points) and I received the following error: Segment violation ('core' generated)

jingnanshi, 2021-07-27:
@jbalado Have you tried adding `OMP_NUM_THREADS=12` to your environment: something like `OMP_NUM_THREADS=12 python ./test.py`?

jbalado, 2021-07-30:
Yes, but the error continues

jingnanshi, 2021-07-30:
Please try to compile everything in debug mode, and use `gdb` to locate the exact line at which the program faults.

LimHyungTae, 2024-04-03:
Yes, I think 4K of correspondences are too many (empirically, TEASER++ works in near-real time under 600 correspondences). Please filter the correspondences before you give them as inputs.

LimHyungTae, 2024-05-02:
If there are no further questions, we will close the issue :) Thx