MIT-SPARK/TEASER-plusplus #112: [ERROR] Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES)

HOUYONGKUO, 2021-08-02:
Hi!  When  running  'cmake -DTEASERPP_PYTHON_VERSION=3.6 .. && make teaserpp_python' in cinnamon system, I got the error:
```
CMake Error at /home/vge/miniconda3/envs/teaser/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES)
Call Stack (most recent call first):
  /home/vge/miniconda3/envs/teaser/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /home/vge/miniconda3/envs/teaser/share/cmake-3.21/Modules/FindOpenMP.cmake:544 (find_package_handle_standard_args)
  build/pmc-src/CMakeLists.txt:26 (find_package)
-- Configuring incomplete, errors occurred!
```
It is grateful if  you can figure it out. Thanks!

HOUYONGKUO, 2021-08-02:
When I changed to a version of pybind11, it prompted:
`corrupted size vs. prev_size
Aborted (core dumped)
`

darkdevahm, 2021-11-17:
@HOUYONGKUO  I have the same problem as (corrupted size vs. prev_size Aborted (core dumped) ), have you reached a solution for this?

jingnanshi, 2021-11-17:
This might be caused by a memory issue --- what example were you running?

darkdevahm, 2021-11-17:
I'm running 3D smooth example using: teaser_python_3dsmooth.py
And this is the full message I get: 
`(teaser_test) ahmed@ahmed:~/ahmed/code/TEASER-plusplus/examples/teaser_python_3dsmooth$ python teaser_python_3dsmooth.py
==================================================
        TEASER++ Python registration example      
==================================================
TEASER++ Parameters are: <Params with noise_bound=0.05
cbar2=1
estimate_sccaling=0
rotation_estimation_algorithm=GNC-TLS
inlier_selection_mode=PMC_EXACT
>
Starting scale solver.
Scale estimation complete.
Max core number: 76
Num vertices: 841
*** [pmc heuristic: thread 16]   current max clique = 61,  time = 5.91278e-05 sec
*** [pmc heuristic: thread 7]   current max clique = 65,  time = 0.00012207 sec
*** [pmc heuristic: thread 2]   current max clique = 66,  time = 0.000171185 sec
[pmc heuristic]	 mc = 66
Created adjacency matrix in 0.000154018 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 840, |E| = 39368
[pmc: initial k-core pruning]  after pruning:  |V| = 840, |E| = 4137
[pmc]  initial pruning took 0.000138044 sec
[pmc: sorting neighbors]  |E| = 8274, |E_sorted| = 8274
|V| = 94
>>> [pmc: thread 16]
[pmc: bounds updated - thread 15]  time = [pmc: bounds updated - thread 0.00067687 sec, |V| = 839 (1 / 840), |E| = 4137, w = 66, p = 0.0117402, d_min = 24, d_avg = 13]  time = 93.7333, d_max = 250, k_max = 76
[pmc: bounds updated - thread 16]  time = 0.000681877 sec, |V| = 830 (10 / 840), |E| = 4137, w = 66, p = 0.000697851 sec, |V| = 0.0117402837, d_min =  (243, d_avg =  / 840), |E| = 4137, w = 66, p = 93.7333, d_max = 250, k_max = 76
0.0117402, d_min = 24, d_avg = 93.7333, d_max = 250, k_max = 76
[pmc: bounds updated - thread 14]  time = 0.00224996 sec, |V| = 828 (12 / 840), |E| = 4137, w = 66, p = 0.0117402, d_min = 24, d_avg = 93.7333, d_max = 250, k_max = 76
-----------------------------------------------------------------------
Max Clique of scale estimation inliers: 
Using chain graph for GNC rotation.
Starting rotation solver.
GNC rotation estimation noise bound:0.1
GNC rotation estimation noise bound squared:0.01
GNC-TLS solver terminated due to cost convergence.
Cost diff: 0
Iterations: 17
Rotation estimation complete.
Starting translation solver.
Translation estimation complete.
Max clique size: 66
double free or corruption (out)
Aborted (core dumped)
`