MIT-SPARK/TEASER-plusplus #113: [QUESTION]

zengletian1491, 2021-09-01:
When I ran the examples of TEASERPlusPlus, it worked very well. 

However, I used TEASERPlusPlus to calculate the rotation matrix and the translation matrix
between two neighboring parts of an object, however, my c++ code ran well but the
results are like in the following: 
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
I guess the parameters about registration solver are not set well.
Could you give some suggestions about how to set the parameters in the following? 
teaser::RobustRegistrationSolver::Params params;
    params.noise_bound = 0.05;
    params.cbar2 = 1;
    params.estimate_scaling = false;
    params.rotation_max_iterations = 100;
    params.rotation_gnc_factor = 1.4;
    params.rotation_estimation_algorithm =
        teaser::RobustRegistrationSolver::ROTATION_ESTIMATION_ALGORITHM::GNC_TLS;
    params.rotation_cost_threshold = 0.005;
    params.inlier_selection_mode = 
        teaser::RobustRegistrationSolver::INLIER_SELECTION_MODE::KCORE_HEU;

What's more, my tested data are uploaded into the attachment. You can modify the .txt to .ply, and you
can load the data using pcl::io:loadPLYFile function. 
[pointcloud_1.txt](https://github.com/MIT-SPARK/TEASER-plusplus/files/7090464/pointcloud_1.txt)
[pointcloud_2.txt](https://github.com/MIT-SPARK/TEASER-plusplus/files/7090469/pointcloud_2.txt)

Thank you very much.

jingnanshi, 2021-09-09:
It is likely that you run out of memory. TEASER++ takes in correspondences, instead of original point clouds. So you need to use a feature matcher to first select correspondences.

LimHyungTae, 2024-05-02:
Yes, 180K points are too many. Empirically, TEASER++ can cover up to < 700-800 points

closed 2024-05-10 by jingnanshi