MIT-SPARK/TEASER-plusplus #115: Support for higher dimensional pointclouds

OlfwayAdbayIgbay, 2021-09-16:
**Have you read the documentation?**

- [ x] Yes

Thank you very much for the great paper and code!

I am working with 3D pointclouds which additionally to the spatial coordinates, also have field properties at every point. I would like to model those properties as extra dimensions.

Does either the python or the C++ bindings support point set registration in higher dimensions? If no, how hard (and how feasible) would it be to modify the code to work for higher dimensions?

jingnanshi, 2021-09-21:
Hi @OlfwayAdbayIgbay, thank you for your interests. Unfortunately we don't have plans to include the functionality for higher dimensional point clouds.

OlfwayAdbayIgbay, 2021-11-05:
Thanks you so much! I've been re-writing some parts of TEASER in python for prototyping, and this code is incredibly helpful! Also, I might be mistaken here, but isn't the C++ code using the Surrogate Geman-McClure distance for rotation rather than the TLS? (see master/teaser/src/registration.cc line 249).

jingnanshi, 2021-11-05:
We have both GM and TLS (https://github.com/MIT-SPARK/TEASER-plusplus/blob/d79d0c678b973e052796bc75d3f77025363726d6/teaser/src/registration.cc#L314)

closed 2022-03-03 by jingnanshi