MIT-SPARK/TEASER-plusplus #117: Scaling seems doesn't work for python

liu-qingzhen, 2021-09-30:
Hi, TEASER++ works great for same scale point clouds with large outliers.  But for point clouds with large scaling difference, the result seems not very good. I'm not sure if it is because the Python binding does not support scaling? I tuned the parameter estimate_scaling in helpers.py to be True, but the performance seems unchanged. 
[image]

liu-qingzhen, 2022-02-18:
>         - External Email -
>
> @liu-qingzhen<https://github.com/liu-qingzhen>, @gaussiangit<https://github.com/gaussiangit> did you manage to have scaling working. I am having a similar scaling use case and the result is not that good as well.
>
>
> Notice: This email is generated from the account of an NUS alumnus. Contents, views, and opinions therein are solely those of the sender.

The result of the model will return the scale. I forgot the code but the scale can be derived from result.s as well as the rotation and translation (result.T).

ttsesm, 2022-02-18:
Sure it returns an `s` value, however it is not correct. When the scaling ratio between the two pcds is quite big, it fails to address the registration and the scaling value doesn't really help much.

In any case, thank you for your time.

liu-qingzhen, 2022-02-19:
Yeah, they said it is because of the feature extraction with fpfh is not accurate. The scale estimation model makes sense, but the performance is not good with traditional feature descriptor.

        - External Email -

ttsesm, 2022-02-21:
I see, did you test with an alternative feature descriptor, other than the fpfh, to see whether it performs better?

jingnanshi, 2022-02-22:
@ttsesm FPFH is not designed to be scale-invariant (see the original paper). Maybe try this: https://link.springer.com/article/10.1007/s00521-017-2964-1?

nkemter, 2026-03-24:
Is this bug/problem specific to Python or does this also occur with MATLAB and C++ implementations?