MIT-SPARK/TEASER-plusplus #119: [QUESTION] 2D alignment

sgrayb, 2021-11-05:
I've been using TEASER to align two 2D point clouds.  The results are not as good as I had hoped after running through the demos and seeing how robust TEASER is to noise and outliers.

In the figure below you'll see four point clouds.  The blue is the source point cloud, the red is the target point cloud, yellow is the target after TEASER alignment, and the purple is target alignment using a ground truth reference system.  Ideally the purple and yellow point clouds would align closely.

I've tried both rotation solvers: GNC_TLS and FGR.  I've tried changing the solver params, e.g., noise_bound, cbar2, etc. with little success.

Is this a matter of tuning TEASER for my dataset or is there something fundamentally about my dataset that causes TEASER to fail?  Also, is there a way to configure TEASER to only do a 2D alignment?  i.e., (to only estimate two translations parameters and one rotation parameter).  The benefits would be compute speed and robustness.

[image]

The solver params are as follows:

```
#define NOISE_BOUND 0.05

teaser::RobustRegistrationSolver::Params params;
params.noise_bound = NOISE_BOUND;
params.cbar2 = 1;
params.estimate_scaling = false;
params.rotation_max_iterations = 100;
params.rotation_gnc_factor = 1.4;
params.rotation_estimation_algorithm =
    teaser::RobustRegistrationSolver::ROTATION_ESTIMATION_ALGORITHM::GNC_TLS;
params.rotation_cost_threshold = 0.005;
```

The output is as follows:

```
Read 15076 total vertices 
Read 1076 total vertices 
Starting TEASER solver
Starting scale solver.
Scale estimation complete.
Max core number: 3
Num vertices: 15077
*** [pmc heuristic: thread 11]   current max clique = 2,  time = 0.000105858 sec
[pmc heuristic]  mc = 2
Created adjacency matrix in 0.0313511 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 15076, |E| = 1587
[pmc: initial k-core pruning]  after pruning:  |V| = 1521, |E| = 191
[pmc]  initial pruning took 0.000163078 sec
[pmc: sorting neighbors]  |E| = 382, |E_sorted| = 382
|V| = 113
-----------------------------------------------------------------------
Max Clique of scale estimation inliers: 
Using chain graph for GNC rotation.
Starting rotation solver.
GNC rotation estimation noise bound:0.1
GNC rotation estimation noise bound squared:0.01
GNC-TLS solver terminated due to cost convergence.
Cost diff: 0.00140892
Iterations: 2
Rotation estimation complete.
Starting translation solver.
Translation estimation complete.
=====================================
          TEASER++ Results           
=====================================
Estimated rotation: 
 0.276848 -0.960914         0
-0.960914 -0.276848        -0
        0        -0        -1

Estimated translation: 
    -2.85733
     29.4295
3.47666e-310
Time taken (s): 3.51892
```

jingnanshi, 2021-11-05:
Did you pass in the correspondences, or the points? 

Currently TEASER++ does not support pure 2D transformation unfortunately.

sgrayb, 2021-11-05:
@jingnanshi Thanks for your reply.  I passed in the points.  But now I understand that I need to also pass in correspondences.  I'm building PCL so that I can run the correspondence estimator (teaser::FPFHEstimation).  I'm not sure how well this will work on my relatively sparse point cloud.

In the "TEASER: Fast and Certifiable Point Cloud Registration" in section XI-C there's some discussion about Simultaneous Pose and Correspondences (SPC).  It says that all possible correspondences are generated.  Is this implemented in the codebase?

Do you know of another algorithm that is likely to be robust with the example point clouds that I show?

jingnanshi, 2021-11-06:
Regarding SPC: yes, and it simply requires you to generate the correspondences yourself. The columns with the same index in `src` and `dst` are the corresponding pair of points. So for SPC, you just need to generate for point i, N-1 possible correspondences. And you concatenate all these points together to be an Eigen matrix.

Regarding algorithms: you can always try different ICP algorithms as they don't require correspondences.

closed 2022-02-22 by jingnanshi

arenas7307979, 2023-12-21:
Hello TEASER++ team,

I've been exploring TEASER++ for point cloud registration and am impressed with its robustness, especially as noted in the paper "TEASER: Fast and Certifiable Point Cloud Registration". The paper mentions that TEASER++ can efficiently solve problems without correspondences, hypothesizing all-to-all correspondences, and often outperforms ICP in such scenarios.

I'm considering using TEASER++ for aligning 2D trajectories. Specifically, I have a 2D vehicle trajectory (src_vdr) and a 2D map of possible vehicle paths (target_roadmap). For both trajectories, the z-axis is set to zero. [input only 2d points] 
[image]

My questions are:
1. Is TEASER++ suitable for aligning such 2D trajectories, especially considering its ability to work without established correspondences?
2. Can TEASER++ be a more accurate alternative to ICP for this kind of 2D point cloud (or trajectory) alignment?
Are there any specific settings or considerations I should be aware of when using TEASER++ for 2D data?

Any insights or recommendations would be greatly appreciated.

Thank you!