MIT-SPARK/TEASER-plusplus #121: [QUESTION] License

patsp, 2021-11-15:
Hi, great work!

It seems that the license of the Parallel Maximum Clique (PMC) library (https://github.com/jingnanshi/pmc) used is GPL 3 (https://github.com/jingnanshi/pmc/blob/master/LICENSE.md), right?

Would it therefore be possible to link against and call your TEASER++ library in for example MIT licensed code? Or would a patch with a compile-time switch to enable/disable an alternative maximum clique solver with conditional double licensing (e.g. MIT and GPL 3) be required?

jingnanshi, 2021-11-15:
Hi, thanks for the question. The PMC library is only downloaded when you run CMake, so technically the GPL 3 license shouldn't be an issue, as the TEASER++ library itself is MIT license, and it becomes GPL v3 only on your machine. You can do something similar on your MIT licensed project.

patsp, 2021-11-15:
Ah, I see, you are right, that makes sense. Thanks for the clarification.

closed 2021-11-15 by patsp

analog-cbarber, 2021-12-15:
Note that the license would be an issue if you were to build a conda or pip distribution that included the pmc component!

analog-cbarber, 2021-12-15:
Really, the MIT license is a bit misleading, because once you compile this code, the binary would fall under the GPL.

I wonder if the PMC author would be open to relaxing his license?

jingnanshi, 2021-12-15:
I messaged the author a few times and he hasn't given me a response. A while ago I started looking into a reimplementation but unfortunately didn't have enough time to carry it forward.