MIT-SPARK/TEASER-plusplus #123: [QUESTION] Using teaser for finding the transformation that connects/merges two point clouds

ttsesm, 2021-12-09:
Hi @jingnanshi,

I've benn using teaser for aligning different point clouds with success. Following the example of using [fpfh+icp](https://github.com/MIT-SPARK/TEASER-plusplus/tree/develop/examples/teaser_python_fpfh_icp) worked quite nicely in my case. Now I wanted to ask whether it would be possible to use the teaser framework to merge two different point clouds. For example look on the two points clouds in the picture below:
[image]
The two pcds connect at a specific part but they are two different instances of a bigger pcd. 

Now I manually applied a transformation to one of the two pieces:
[image]
and I would like to find the transformation that does not align but rather connects the two. Would that be possible with teaser?

I've continued on the suggested steps from the aforementioned example and I could see that the fpfh can extract meaningful correspondences:
[image]
and that teaser+icp can correctly identify the side that the two pcds have the similarities:
[image]
Though in reverse format. Thus, the extracted transformation is not correct since it tries to match similar points and not similar points in reverse. I was thinking whether I could play somehow either with some of the teaser of the fpfh parameter for adjusting the correspondence points and how these match to each other.

Thanks.

jingnanshi, 2021-12-10:
Hi @ttsesm, thank you for your interest. The point clouds you are showing here are unfortunately very difficult to "merge" in the way you want it, as the geometry are a bit repetitive and there might exists multiple possible registration solutions. 

Do you have access to any additional information regarding the relative positions between the two point clouds?

ttsesm, 2021-12-10:
> Hi @ttsesm, thank you for your interest. The point clouds you are showing here are unfortunately very difficult to "merge" in the way you want it, as the geometry are a bit repetitive and there might exists multiple possible registration solutions.
> 
It is true that there are multiple registration solutions, however from the example I can see that the correspondence points are kind of pointing to the correct side. Plus if I isolate the area that the two point clouds seem to merge I can see that the points and the geometry structure that they create is completing each other:
[image]
Thus there should be a way to take advantage of this feature.

> Do you have access to any additional information regarding the relative positions between the two point clouds?
>
No, unfortunately the point clouds are randomly positioned in the space and then I am trying to find the best assembly solution. The only additional information that I could possibly make use is the texture which I have available.

jingnanshi, 2021-12-10:
Very interesting. Usually in the existing works of point cloud registration, people find correspondences from the same points in two scans (i.e., there are overlapping regions in the two point clouds). However, in your case you are actually looking for correspondences that make the scans fit like a jigsaw puzzle. I am not confident that existing hand-crafted feature extractors will work well, given that they find points in the feature space by looking at similar local geometries, whereas in your case you need to find points with complementary local geometries.

If you can generate synthetic data easily, I suggest training a point cloud encoder-decoder network.

closed 2022-02-22 by jingnanshi