MIT-SPARK/TEASER-plusplus #127: Must I use FPFH first?

zikai1, 2022-03-02:
**Have you read the documentation?**

- [ *]  Yes

- [ ] No - then this issue will be closed.

Hi Jingnan,
  thanks for the excellent work. I have compiled the code and successfully run it. I find in the directory **examples**, there are three test cases. The first is based on the TLS without point features: **teaser_python_ply** and **teaser_cpp_ply**, the second is based on the FPFH feature: **teaser_python_fpfh_icp** and **teaser_cpp_fpfh**, and the last one is based on the learning feature: **teaser_python_3dsmooth**. I used two point clouds with outliers and ran **teaser_python_ply**, but the result is unsatisfactory. However, **teaser_python_fpfh_icp** can produce satisfactory results. My question is that must I use fpfh feature for all cases in advance? How **teaser_python_ply** return certifiable registration under outlier contamination? Thanks.

jingnanshi, 2022-03-02:
No, you can use any feature detector you want. The solve() function just takes in corresponding points as two matrices. The certifier currently only have the C++ API, so you might need to use C++.

zikai1, 2022-03-03:
Hi Jingnan,
thanks for your reply. If I directly input two point clouds without feature, also without advanced correspondence, like the simplest ICP, can I still get certible results? Thanks.

zikai1, 2022-03-04:
> You need to formulate correspondences first. So if you decide to not use a feature detector, then you need to feed all the possible correspondences (N*(N-1)/2 where N is the number of points) into the solver. ICP can be replicated by creating correspondences by minimizing pairwise distances, but the solution might not be as good as that from using a feature detector first.

Hi Jingnan,
Thanks. I know how to produce good registration results. Thank you very much.

jingnanshi, 2022-03-04:
Thank you for your interest! Closing this issue now.

closed 2022-03-04 by jingnanshi