MIT-SPARK/TEASER-plusplus #129: [BUG] Memory release errors during solving

AndreiCostinescu, 2022-05-08:
**Describe the bug**
I get memory allocation errors from the ```solver.solve(x, y)``` function (```munmap_chunk(): invalid pointer``` or ```double free or corruption (out)```) when running a simple registration example (code and data attached, see To Reproduce section).

**Have you run the unit tests?**
Test project /usr/local/src/TEASER-plusplus/build-release
      Start  1: UtilsTest.RandomSample
 1/48 Test  1: UtilsTest.RandomSample ...............................   Passed    0.00 sec
      Start  2: UtilsTest.EigenMatrixRemoveRowColumn
 2/48 Test  2: UtilsTest.EigenMatrixRemoveRowColumn .................   Passed    0.00 sec
      Start  3: UtilsTest.CalculatePointClusterDiameter
 3/48 Test  3: UtilsTest.CalculatePointClusterDiameter ..............   Passed    0.00 sec
      Start  4: IOTest.ImportPLY
 4/48 Test  4: IOTest.ImportPLY .....................................   Passed    0.00 sec
      Start  5: IOTest.ImportBigPLY
 5/48 Test  5: IOTest.ImportBigPLY ..................................   Passed    0.22 sec
      Start  6: PointXYZTest.SimpleOperations
 6/48 Test  6: PointXYZTest.SimpleOperations ........................   Passed    0.00 sec
      Start  7: PointCloudTest.SimpleOperations
 7/48 Test  7: PointCloudTest.SimpleOperations ......................   Passed    0.00 sec
      Start  8: TLSTest.TLSEstimate
 8/48 Test  8: TLSTest.TLSEstimate ..................................   Passed    0.00 sec
      Start  9: TLSTest.TLSEstimateTiled
 9/48 Test  9: TLSTest.TLSEstimateTiled .............................   Passed    0.01 sec
      Start 10: ScaleSolverTest.UnknownScale
10/48 Test 10: ScaleSolverTest.UnknownScale .........................   Passed    0.00 sec
      Start 11: ScaleSolverTest.FixedScale
11/48 Test 11: ScaleSolverTest.FixedScale ...........................   Passed    0.00 sec
      Start 12: RotationSolverTest.FGRRotation
12/48 Test 12: RotationSolverTest.FGRRotation .......................   Passed    0.00 sec
      Start 13: RotationSolverTest.GNCTLS
13/48 Test 13: RotationSolverTest.GNCTLS ............................   Passed    0.00 sec
      Start 14: TranslationTest.TLSTranslation
14/48 Test 14: TranslationTest.TLSTranslation .......................   Passed    0.00 sec
      Start 15: RegistrationTest.LargeModel
15/48 Test 15: RegistrationTest.LargeModel ..........................   Passed    0.01 sec
      Start 16: RegistrationTest.SolveForScale
16/48 Test 16: RegistrationTest.SolveForScale .......................   Passed    0.01 sec
      Start 17: RegistrationTest.SolveForRotation
17/48 Test 17: RegistrationTest.SolveForRotation ....................   Passed    0.00 sec
      Start 18: RegistrationTest.SolveRegistrationProblemDecoupled
18/48 Test 18: RegistrationTest.SolveRegistrationProblemDecoupled ...***Exception: SegFault  0.08 sec
      Start 19: RegistrationTest.OutlierDetection
19/48 Test 19: RegistrationTest.OutlierDetection ....................   Passed    0.01 sec
      Start 20: RegistrationTest.NoMaxClique
20/48 Test 20: RegistrationTest.NoMaxClique .........................   Passed    0.00 sec
      Start 21: RegistrationTest.CliqueFinderModes
21/48 Test 21: RegistrationTest.CliqueFinderModes ...................   Passed    0.00 sec
      Start 22: GraphTest.BasicFunctions
22/48 Test 22: GraphTest.BasicFunctions .............................   Passed    0.00 sec
      Start 23: PMCTest.FindMaximumClique1
23/48 Test 23: PMCTest.FindMaximumClique1 ...........................   Passed    0.01 sec
      Start 24: PMCTest.FindMaximumClique2
24/48 Test 24: PMCTest.FindMaximumClique2 ...........................   Passed    0.01 sec
      Start 25: PMCTest.FindMaximumClique3
25/48 Test 25: PMCTest.FindMaximumClique3 ...........................   Passed    0.01 sec
      Start 26: MaxCliqueSolverTest.FindMaxClique
26/48 Test 26: MaxCliqueSolverTest.FindMaxClique ....................   Passed    0.01 sec
      Start 27: DRSCertifierTest.GetOmega1
27/48 Test 27: DRSCertifierTest.GetOmega1 ...........................   Passed    0.02 sec
      Start 28: DRSCertifierTest.GetBlockDiagOmega
28/48 Test 28: DRSCertifierTest.GetBlockDiagOmega ...................   Passed    0.02 sec
      Start 29: DRSCertifierTest.GetQCost
29/48 Test 29: DRSCertifierTest.GetQCost ............................   Passed    0.02 sec
      Start 30: DRSCertifierTest.GetLambdaGuess
30/48 Test 30: DRSCertifierTest.GetLambdaGuess ......................   Passed    0.02 sec
      Start 31: DRSCertifierTest.GetLinearProjection
31/48 Test 31: DRSCertifierTest.GetLinearProjection .................   Passed    0.02 sec
      Start 32: DRSCertifierTest.GetOptimalDualProjection
32/48 Test 32: DRSCertifierTest.GetOptimalDualProjection ............   Passed    0.03 sec
      Start 33: DRSCertifierTest.ComputeSubOptimalityGap
[... 114 lines cut ...]
|V| = 270
>>> [pmc: thread 16]
[pmc: bounds updated - thread 13]  time = [pmc: bounds updated - thread 0.000206947 sec, |V| = 290 (117 / 407), |E| = 2793, w = 10, p = 0.03380514, d_min = ]  time = 1, d_avg = 16.9287, d_max = 33, k_max = 21
0.000213861 sec, |V| = 286 (121 / 407), |E| = 2793, w = 10, p = 0.033805, d_min = 1, d_avg = 16.9287, d_max = 33, k_max = 21
[pmc: bounds updated - thread 16]  time = 0.000232935 sec, |V| = 291 (116 / 407), |E| = 2793, w = 10, p = 0.033805, d_min = 1, d_avg = 16.9287, d_max = 33, k_max = 21
[pmc: bounds updated - thread 15]  time = 0.00169587 sec, |V| = 274 (133 / 407), |E| = 2793, w = 10, p = 0.033805, d_min = 1, d_avg = 16.9287, d_max = 33, k_max = 21
-----------------------------------------------------------------------
Max Clique of scale estimation inliers: 
Using chain graph for GNC rotation.
Starting rotation solver.
GNC rotation estimation noise bound:0.02
GNC rotation estimation noise bound squared:0.0004
GNC-TLS solver terminated due to cost convergence.
Cost diff: 0
Iterations: 14
Rotation estimation complete.
Starting translation solver.
Translation estimation complete.
Found solution? 1
t = 103.638 -153.34 106.407
R =
-1.54759e-05    -0.511897    -0.859047
          -1  1.54759e-05  8.79329e-06
 8.79329e-06     0.859047    -0.511897
s = 1

lidonghui-ai, 2022-05-27:
Have you solved this bug yet?
The same bug was encountered

AndreiCostinescu, 2022-05-27:
No, this is still an active issue for me...

lidonghui-ai, 2022-05-28:
> No, this is still an active issue for me...
Hi，I have solve it by setting the environment variable OMP_NUM_THREADS=12.
export OMP_NUM_THREADS=12

lidonghui-ai, 2022-05-28:
see issue https://github.com/MIT-SPARK/TEASER-plusplus/issues/54#issuecomment-829305428 for details.

> > No, this is still an active issue for me...
> 
> Hi，I have solve it by setting the environment variable OMP_NUM_THREADS=12. export OMP_NUM_THREADS=12

jingnanshi, 2022-07-04:
@AndreiCostinescu @lidonghui-ai In the latest commit to the `develop` branch I have fixed this issue. Now you should be able to run it without setting the environmental variable. Please let me know if you run into any more issues.

LimHyungTae, 2024-05-02:
If there are no further questions, we will close the issue :) Thx

closed 2024-05-18 by jingnanshi