MIT-SPARK/TEASER-plusplus #130: [QUESTION] test on my own datasets, registration failed

zener90818, 2022-05-12:
here the results，how to adapt the code on my own datasets, thansk a lot.
[image]
[image]
[image]

jingnanshi, 2022-05-13:
What feature extraction algorithm did you use?

zener90818, 2022-05-14:
> 

the FPFH features are used.
[image]

jingnanshi, 2022-05-14:
Can you try registering your source with itself using FPFH features?

zener90818, 2022-05-15:
> Can you try registering your source with itself using FPFH features?

I have tried two different way to register. 
(1) From source to itself, registration success,
[image]
(2) from source to itself  which is applied with a random rotation and translation matrix,  registration failed,
[image]
[image]

jingnanshi, 2022-05-15:
Do you mind posting the entire code snippet you used for doing the self-registration experiment? My guess is the correspondence-matching part is causing this issue.

LimHyungTae, 2024-05-02:
If there are no further questions, we will close the issue :) Thx

closed 2024-05-18 by jingnanshi