MIT-SPARK/TEASER-plusplus #131: [BUG] When I run teaser_cpp_fpfh example,I run into this problem

nottoday0927, 2022-05-24:
Thanks for your work.!!!I followed the steps to compile Python binding, everything went well, but when I run teaser_python_ply.py or other examples, the following error appears, how can I solve it?

**Describe the bug**
ModuleNotFoundError: No module named 'teaserpp_python.teaserpp_python'

**Have you run the unit tests?**

- [ ] Yes

**Installed Dependencies**
ubuntu 20.01

**To Reproduce**
(teaser_test) root@yjh-virtual-machine:/home/yjh/TEASER-plusplus/examples/teaser_python_ply# python teaser_python_ply.py

Traceback (most recent call last):
  File "teaser_python_ply.py", line 5, in <module>
    import teaserpp_python
  File "/home/yjh/anaconda3/envs/teaser_test/lib/python3.6/site-packages/teaserpp_python/__init__.py", line 1, in <module>
    from .teaserpp_python import *
ModuleNotFoundError: No module named 'teaserpp_python.teaserpp_python'

gitouni, 2022-05-25:
Have you build your teaserpp_python module with pybind11?

nottoday0927, 2022-06-15:
> Have you built your teaserpp_python module with pybind11?
Thank you for answering patiently！ I have solved the problem

closed 2022-06-15 by nottoday0927

SMU-Terry, 2023-06-21:
I met the same error, may you share the experience of fixing it? thx so much!