MIT-SPARK/TEASER-plusplus #134: [QUESTION] Translation only alignment?

cclaan, 2022-06-26:
**Have you read the documentation?**

- [x] Yes
- [] No 

Hi, for the case of 3dof translation point cloud alignment, are there modifications to teaser that would improve speed / accuracy over just running the full rotation/scale/translation steps?  
Thanks

jingnanshi, 2022-07-01:
You should be able to just run the translation estimator, although you need to take a look into the implementation as we don't have an existing API for translation only estimation.

cclaan, 2022-07-01:
Thanks,
I did take a look at the translation estimation step, but it looked like the point clouds had already been pruned from the rotation / scale parts that use the correspondences. There was no way to pass correspondences directly to the translation step.

cclaan, 2022-07-05:
Thanks I will dig in a bit further

closed 2022-07-05 by cclaan