MIT-SPARK/TEASER-plusplus #135: [QUESTION] How to set OMP_NUM_THREADS=12 for Matlab in Ubuntu

WulongGuo, 2022-06-29:
**Have you read the documentation?**

- [ ] Yes
- [x] No - then this issue will be closed.

Post your theoretical questions / usage questions here.
Dear author,
Thank you for your code. 
I know how to set OMP_NUM_THREADS=12 in a C++ code, and it surely will avoid the PMC crash problem. Now I want to use teaser++ in matlab with the matlab bindings. is there a way to set OMP_NUM_THREADS=12 for matlab? I have tried using "set OMP_NUM_THREADS=12" in a Ubuntu command shell before opening matlab, but it didn't work.

jingnanshi, 2022-07-03:
Try the latest develop branch. I fixed the issue with thread count issue so you don't have to set the environment variable. Make sure to clean all the CMake build folders, reload CMake project and rebuild.

WulongGuo, 2022-07-03:
Thank you! I am not sure if I build it right or not, but when the number of points gets over 10000, Matlab would  crash. I will use the C++ code, it works good for me. Thanks again.

jingnanshi, 2022-07-04:
Yes due to the implementation the memory consumption will be large with high number of points/correspondences.