MIT-SPARK/TEASER-plusplus #139: [QUESTION] The result is not ideal in the example teaser_cpp_fpfh

HarborC, 2022-07-21:
**Have you read the documentation?**

- [x] Yes
- [] No - then this issue will be closed.

Post your theoretical questions / usage questions here.
[image]

Thank you for your excellent open source work. I have a question for you. I tried to run the example teaser_cpp_fpfh  and found that the result was not ideal. Do you also find such a problem?

stephen9999999, 2022-08-20:
Same problem. have you solved it?

LimHyungTae, 2024-04-03:
What do you mean "ideal"? Do you mean the error is not equal to zero? It's the natural characteristic of a non-minimal solver. Because the problem we want to solve does not satisfy the following condition: actual global optimum != global optimum in the objective function. For this reason, the result is near-zero, but not zero.

LimHyungTae, 2024-05-02:
In short, it does not go to global optimum to reduce the effect of outliers (because owing to the outliers, the actual global optimum != global optimum in the loss function space). If there are no further questions, we will close the issue :) Thx

closed 2024-05-10 by jingnanshi