MIT-SPARK/TEASER-plusplus #141: [QUESTION]Inconsistent results between c++ and python implementation

stephen9999999, 2022-08-19:
**Have you read the documentation?**

- Yes

Post your theoretical questions / usage questions here.

Hi, thanks for your great work.
I have test teaser_cpp_fpfh and teaser_python_fpfh_icp with the same pcd files, but the results are inconsistent. Is there any difference of this two implementation?

For teaser_python_fpfh_icp, I test the algorithm with two pcd files collected by myself. And I get the reasonable result. The estimated transforms are:
T_teaser:  
[[ 0.93723273  0.11378536 -0.3296175  -0.09877629]
 [-0.10340515  0.99343573  0.04891655  0.02664667]
 [ 0.33301979 -0.01176205  0.94284647  0.10280838]
 [ 0.          0.          0.          1.        ]]

T_icp_refine:
 [[ 0.93582733 -0.00977672 -0.35232319  0.10150537]
 [ 0.02057001  0.99942636  0.02690389  0.02983166]
 [ 0.35185805 -0.03242469  0.93549161  0.08510608]
 [ 0.          0.          0.          1.        ]]

For teaser_cpp_fpfh, I slightly modify the code to input two pcd files. And the estimated transform is:
Estimated rotation: 
 0.341367 -0.376026  0.861437
 0.451216  0.869541  0.200758
-0.824544  0.320162  0.466501
Estimated translation: 
 -4.24419
-0.724059
   2.3231
The estimated transform is not reasonable.

May I ask why the result of c++ implementation is not reasonable?  how can I improve the result of c++ version?

jingnanshi, 2022-08-19:
Make sure in the C++ implementation to call reset after every solve using the solver.

stephen9999999, 2022-08-20:
Thanks for your reply.
In the demo teaser_cpp_fpfh.cc, the solver is called only once. I input tow pcd files, and use the solver to calculate the transform of the two point cloud.
I follow your suggestion and call reset after every solve

[image]

but I get an error: "error: no matching function for call to ‘teaser::RobustRegistrationSolver::reset()’".
what should I do to improve the result of c++ implementation? looking forward to your suggestion.

jingnanshi, 2022-08-21:
See this line: https://github.com/MIT-SPARK/TEASER-plusplus/blob/111e2ce81aa8c25fad66fcd1feecb758e1c76281/teaser/include/teaser/registration.h#L738

Can you also try using `solve(src_correspondences, tgt_correspondences)` instead of passing the correspondences in separately? Basically just make sure the columns of `src_correspondences` and `tgt_correspondences` are corresponding 3D points.

LimHyungTae, 2024-05-02:
If there are no further questions, we will close the issue :) Thx

closed 2024-05-02 by jingnanshi