MIT-SPARK/TEASER-plusplus #144: [BUG] scale estimation cannot be forbiddened in python

gitouni, 2022-09-05:
**Describe the bug**
A clear and concise description of what the bug is.

**Have you run the unit tests?**

- [x] Yes
- [ ] No --- then this issue will be closed.

I have passed estimated_scale=False to teaser solver but it solves scale estimation still.

**To Reproduce**
Teaser++ solver class:
```py
import teaserpp_python
import numpy as np

class TEASER:
    def __init__(self,noise_bound=0.01):
        solver_params = teaserpp_python.RobustRegistrationSolver.Params()
        solver_params.cbar2 = 1.0
        solver_params.noise_bound = noise_bound
        solver_params.estimate_scaling = False
        solver_params.inlier_selection_mode = \
            teaserpp_python.RobustRegistrationSolver.INLIER_SELECTION_MODE.PMC_EXACT
        solver_params.rotation_tim_graph = \
            teaserpp_python.RobustRegistrationSolver.INLIER_GRAPH_FORMULATION.CHAIN
        solver_params.rotation_estimation_algorithm = \
            teaserpp_python.RobustRegistrationSolver.ROTATION_ESTIMATION_ALGORITHM.GNC_TLS
        solver_params.rotation_gnc_factor = 1.4
        solver_params.rotation_max_iterations = 10000
        solver_params.rotation_cost_threshold = 1e-16
        self.solver_params = solver_params

    def register_based_on_correspondence(self, src:np.ndarray,dst:np.ndarray)-> np.ndarray:
        # src, dst (3,N)
        solver = teaserpp_python.RobustRegistrationSolver(self.solver_params)
        solver.solve(src.astype(np.float64).T,dst.astype(np.float64).T)  # (3,N1), (3,N2) must be double vectors
        solution = solver.getSolution()
        est_g = np.eye(4)
        est_g[:3,3] = solution.translation
        est_g[:3,:3] = solution.rotation
        return est_g
```
TEASER++ solver use:
```py
def teaser_register(self, xyz0, xyz1):
    '''
    Main algorithm of TEASER++ Registration
    '''
   ...
    T = self.teaser_solver.register_based_on_correspondence(xyz0[corres_idx0],xyz1[corres_idx1])  # two point clouds
    ...
```
some output are shown as follows:
```
Starting scale solver.                                                                                                                                                           
Scale estimation complete.
Max core number: 217
Num vertices: 37720
*** [pmc heuristic: thread 1]   current max clique = 6,  time = 0.00213408 sec
*** [pmc heuristic: thread 1]   current max clique = 16,  time = 0.00304198 sec
*** [pmc heuristic: thread 4]   current max clique = 17,  time = 0.0036931 sec
```

**Additional context**
Sometimes, it takes minutes to estimate scale without any output.

jingnanshi, 2022-10-03:
@gitouni Hi, sorry for the confusion. If you disable scale estimation in the params, the solver won't estimate scale. However, it will still use scale as a method to remove outliers. That part might take some time, especially given many correspondences.

gitouni, 2022-10-03:
> @gitouni Hi, sorry for the confusion. If you disable scale estimation in the params, the solver won't estimate scale. However, it will still use scale as a method to remove outliers. That part might take some time, especially given many correspondences.

Thank you for your reply, it becomes more explanable and compatibale to the paper.