MIT-SPARK/TEASER-plusplus #145: [BUG]

adam-muk, 2022-09-07:
**Describe the bug**
CMake Error at CMakeLists.txt:7 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/teaserpp/teaserppConfig.cmake

  but it set teaserpp_FOUND to FALSE so package "teaserpp" is considered to
  be NOT FOUND.  Reason given by package:

  The following imported targets are referenced, but are missing: tinyply
  pmc::pmc

**Have you run the unit tests?**

- [x] Yes - all passed
- [] No --- then this issue will be closed.

**Installed Dependencies**
Ubuntu 20.04 
I followed readme instructions for all dependencies

**To Reproduce**
sudo apt install cmake libeigen3-dev libboost-all-dev
git clone https://github.com/MIT-SPARK/TEASER-plusplus.git
cd TEASER-plusplus && mkdir build && cd build
cmake .. && make
sudo make install
sudo ldconfig
cd .. && cd examples/teaser_cpp_ply && mkdir build && cd build
cmake .. && make
./teaser_cpp_ply

jingnanshi, 2022-10-02:
@adam-muk During which step did you encounter the error message posted? Thanks.

adam-muk, 2022-10-09:
Hi,

Sorry for the delayed response. I reinstalled ubuntu and now I do not get the error anymore. I am not sure what the problem was :/

closed 2022-10-09 by adam-muk

Zedzou, 2023-03-23:
I have this problem too, Any ideas to solve it ?

Zedzou, 2023-03-23:
> I have this problem too, Any ideas to solve it ?
oh. add "find_dependency(tinyply REQUIRED)" and "find_dependency(pmc REQUIRED)" in cmake/teserppConfig.cmake, and it successed.