MIT-SPARK/TEASER-plusplus #146: [QUESTION] Is it possible to disable scaling, rotation or translation estimation?

Scyten, 2022-09-16:
**Have you read the documentation?**

- [x] Yes
- [] No - then this issue will be closed.

Is it possible to disable scaling, rotation or translation estimation?

I have a time critical problem and need only the translation estimation, rotation is 0 and scaling is 1.
Is there a way to disable the scaling and rotation estimation?

If I use `solve.solveForTranslation(Matrix3Xd&, Matrix3Xd&)`, there is no filtering of correspondences, right?

jingnanshi, 2022-10-02:
@Scyten Hi, the scale estimation option will disable the scale solver, but it will still try to filter out outliers using scale. Unfortunately we don't have a way to disable rotation. If you use the estimate translation function directly, it will still have inlier selection based on TLD cost.

jingnanshi, 2022-10-02:
Alternatively, if all you need are translations, perhaps writing a custom solver with the max clique outlier rejection will be a better option.

LimHyungTae, 2024-05-02:
If there are no further questions, we will close the issue. Thx!

closed 2024-05-10 by jingnanshi