MIT-SPARK/TEASER-plusplus #147: [QUESTION]

pahoffmann, 2022-10-02:
**Have you read the documentation?**

- [x] Yes
- [] No - then this issue will be closed.

Post your theoretical questions / usage questions here.

Some background to my question:
I am currently investigating loop closure's in a graph based SLAM and
need to do a scan-matching between two laserscans.
For each of the scans, there exists an initial pose estimate. The combination of the two scans and the two belonging poses
is estimated to be a loop closure candidate.
To verify the candidate i apply a scan matching algorithm like ICP/GICP and evaluate the respective fitness scores.
Though this produces sufficient results in some cases, a better scan matching technique is needed.
This is where teaser-plusplus comes in handy.
I want to use teaser-plusplus to produce a good initial estimate between the scans and apply a post-registration with
ICP afterwards, as suggested.

To get an initial hang of it, i took a look at the **teaser_cpp_fpfh** example, which this question is all about:
When building and executing the example, i get the following output for the registration result:
(notice i added a method to make the expected and estimated transform more human readable):

`=====================================
          TEASER++ Results           
=====================================
Expected rotation: 
  0.996927  0.0668736 -0.0406664
 -0.066129   0.997618  0.0194009
 0.0418676 -0.0166518   0.998978
Pos:    -0.12 | -0.04 | 0.11
Angles: 3.12 | -3.10 | 3.07
Estimated rotation: 
 0.76 -0.08 -0.64
 0.28 -0.86  0.44
-0.59 -0.51 -0.63
Error (deg): 2.62
Pos:    -0.10 | 0.11 | 0.14
Angles: 0.61 | -2.44 | -3.04

Expected translation: 
-0.12
-0.04
 0.11
Estimated translation: 
-0.10
 0.11
 0.14
Error (m): 0.15

Number of correspondences: 1889
Number of outliers: 1700
Time taken (s): 0.00`

As you can clearly see, teaser-plusplus does not even converge in the given example, but terminates early on, because
_GNC-TLS terminated because maximum residual at initialization is very small._.
I am not sure if this is expected behavior, but the algorithm still proceeds to calculate the final transformation,
which is also completely false (The cloud gets rotated by an average of 3 degrees and the resulting error in the rotation is also almost at 3 degrees - so there is an error of almost 100%).
I am pretty sure, that this is not the expected behavior for this test case and might need some looking into.
If i am completely wrong, please enlighten me! :)

Cheers

jingnanshi, 2022-10-02:
@pahoffmann Which branch did you run the code on?

pahoffmann, 2022-10-02:
On the master branch! should i use the develop instead?

jingnanshi, 2022-10-02:
Please try the develop again. A while ago I remember I updated a few things regarding FPFH on develop. I'm also looking into this right now. Also, please run `ctest` to run all unit tests to ensure everything passes (on develop). Thanks!

pahoffmann, 2022-10-02:
On it!

pahoffmann, 2022-10-02:
@jingnanshi
I just completely rebuild based on the develop and actually a lot of unit tests fail:

The following tests FAILED:
	 15 - RegistrationTest.LargeModel (SEGFAULT)
	 16 - RegistrationTest.LargeModelSingleThreaded (SEGFAULT)
	 17 - RegistrationTest.SolveForScale (SEGFAULT)
	 18 - RegistrationTest.SolveForRotation (SEGFAULT)
	 19 - RegistrationTest.SolveRegistrationProblemDecoupled (Child aborted)
	 20 - RegistrationTest.OutlierDetection (SEGFAULT)
	 21 - RegistrationTest.NoMaxClique (Child aborted)
	 22 - RegistrationTest.CliqueFinderModes (Child aborted)
	 46 - RegistrationBenchmark.Benchmark1 (Child aborted)
	 47 - RegistrationBenchmark.Benchmark2 (Child aborted)
	 48 - RegistrationBenchmark.Benchmark3 (Child aborted)
	 49 - RegistrationBenchmark.Benchmark4 (Child aborted)
	 50 - RegistrationBenchmark.Benchmark5 (Child aborted)
	 51 - RegistrationBenchmark.Benchmark6 (SEGFAULT)

Mostly due to segfaults.

I also executed the FPFH example which yields worse results than on the master branch.

pahoffmann, 2022-10-02:
On master, all the tests pass without any problem.

jingnanshi, 2022-10-02:
> On master, all the tests pass without any problem.

Can you check what commit the PMC library downloaded by CMake is? Also, try run ctest with 'OMP_NUM_THREADS=12 ctest'

pahoffmann, 2022-10-02:
The PMC Library (pmc-src) is up to date with It's master branch. Running the tests with above commands yields the same errors.

jingnanshi, 2022-10-03:
Updates on this issue: I can reproduce the issue with the FPFH example. I suspect the issue was introduced when I changed the FPFH class to use the PCL's multi-threaded implementation. `teaser_cpp_ply` example works fine so the problem is limited in the feature matcher.

jingnanshi, 2022-10-03:
I cannot seem to reproduce the seg-faults in the unit tests, however.

pahoffmann, 2022-10-03:
Thank you so much for looking into this!

I am gonna look into the feature matching today.

pahoffmann, 2022-10-03:
Those segfaults are giving me an headache. What key differences are there between master and develop?
Just overall more parallelized? Or actually some extensive changes in the algorithms?

jingnanshi, 2022-10-27:
@pahoffmann If you are still working on this, do you mind try this branch's FPFH example: https://github.com/MIT-SPARK/TEASER-plusplus/tree/bugfix/examples ? Thanks!

TongxingJin, 2024-04-01:
> @pahoffmann If you are still working on this, do you mind try this branch's FPFH example: https://github.com/MIT-SPARK/TEASER-plusplus/tree/bugfix/examples ? Thanks!

I solved the segment fault by keeping BUILD_WITH_MARCH_NATIVE false.

closed 2024-05-10 by jingnanshi