MIT-SPARK/TEASER-plusplus #151: [QUESTION] Feature points selection.

RyanAnvie, 2022-11-04:
**Have you read the documentation?**

- [x] Yes
- [ ] No - then this issue will be closed.

Post your theoretical questions / usage questions here.

Hi can you explain a little bit explain how you select feature points for registration or are you using entire inlier points to match the two-point sets?

weterking, 2022-11-09:
Although I am not the author, but I read the source code, the author seems to be using the FGR method of selecting corresponding points, using the corresponding points initially selected by FGR to perform TEASER

jingnanshi, 2022-11-09:
@RyanAnvie Thanks for your interest. @weterking is correct, the matching code is based on FGR. However, the matcher implemented in this repo is not actually used in the paper. The matcher in this repo is served as a general method for users. For the original paper, we just used a k-NN tree to match feature correspondences (based on 3DSmoothNet).

LimHyungTae, 2024-04-03:
Yeah, it's presented in Section 3.3 of the original FGR paper: see Section 3.3 in https://vladlen.info/papers/fast-global-registration.pdf

LimHyungTae, 2024-05-02:
If there are no further questions, we will close the issue :) Thx

closed 2024-05-02 by jingnanshi