MIT-SPARK/TEASER-plusplus #156: [BUG] Problem Building Python Bindings

LemonPi, 2022-11-25:
**Describe the bug**
Following the minimal python readme steps exactly, I'm getting the following error when trying to run the example:
```
  File "teaser_python_ply.py", line 5, in <module>
    import teaserpp_python
  File "/home/zhsh/miniconda3/envs/teaser_test/lib/python3.6/site-packages/teaserpp_python/__init__.py", line 1, in <module>
    from .teaserpp_python import *
ModuleNotFoundError: No module named 'teaserpp_python.teaserpp_python'
```

I've built the c++ version and it works fine on the example, and in the output after `cmake ...` I see that python bindings have been built:
```
[100%] Built target pmc
-- Found PythonInterp: /usr/bin/python (found suitable version "3.8.10", minimum required is "3.6") 
-- pybind11 v2.4.3
-- TEASER++ Python binding will be built.
-- Python Interpreter Version: 3.8
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zhsh/catkin_ws/src/TEASER-plusplus/build
[ 60%] Built target pmc
[ 86%] Built target teaser_registration
[100%] Built target teaserpp_python
```

I've also read the previous related issues #131 and #92 

**Installed Dependencies**
Ubuntu 20.04
cmake version 3.16.3
conda 4.13.0

**To Reproduce**
Execute the commands
```
sudo apt install cmake libeigen3-dev libboost-all-dev
conda create -n teaser_test python=3.6 numpy
conda activate teaser_test
conda install -c open3d-admin open3d=0.9.0.0
git clone https://github.com/MIT-SPARK/TEASER-plusplus.git
cd TEASER-plusplus && mkdir build && cd build
cmake -DTEASERPP_PYTHON_VERSION=3.6 .. && make teaserpp_python
cd python && pip install .
cd ../.. && cd examples/teaser_python_ply 
python teaser_python_ply.py
```

ScarWar, 2023-02-08:
Any updates on the issue? I would love to help if I can

EdoardoPassera, 2023-04-02:
I have the same issue, did you resolve it?

jingnanshi, 2023-04-03:
@EdoardoPassera can you try the latest develop branch? Thanks

EdoardoPassera, 2023-04-04:
I still get the same error.

Asherchi, 2023-09-20:
me too, did any handsome guy solved this problems?

HikariS97, 2023-12-05:
I have met same problem under Window 10 MinGW toolchain. Here's some steps I took to make it work.

1. Under Windows, pybind11 tends to generate .pyd instead of .so, we have to change the line 11 at ${ProjectRoot}/python/setup.py.in, from `package_data={'': ['*.so]}` to `package_data={'': ['*.pyd']}`. This step make sure that the compiled *.pyd or *.so file can be copied to pip installation path.

2. Make sure that the FOUND PYTHON INTERPRETER is the one as the same as you use for running py scripts. Otherwise, `import` can't find the entry of the Library. Pay attention if you use Anaconda Env, pybind11 may use base python.exe, but you may run scripts in your established envs.

3. Copy the generated *.dll or *.so file**s** under ${ProjectRoot}/build/teaser, and other path (e.g. pmc-build, tinyply-build) to the same path where the py scripts placed, in case that the searching path for dynamic load libraries is not correctly configured.

harshap-ai, 2025-03-07:
During installation (https://teaser.readthedocs.io/en/master/installation.html), install pybind11 before running `cmake ..` from the build folder. This helps make sure that the python interpreter path points to the desired environment in which you're working rather than the generic `/usr/bin/python`.