MIT-SPARK/TEASER-plusplus #162: [BUG] Open3D version incompatibly

adrienzhh, 2023-02-10:
I was trying to run the gif python example from the main page. I followed the shell instruction and installed 0.9.0 open3d version. This would raise o3d has no pipelines attributes error. I was able to solve this one by simply erased the pipelines feature. However, this then later raise aborted (core dumped) since the feature feed in seems to still require pipelines. I solved it using 0.11.0 open3d. Is this a common problem or maybe we have to update README to a new open3d version. Thanks.

EdoardoPassera, 2023-04-03:
I have the core dumped problem, maybe I need to use o3d 0.11.0.

abhishekmonogram, 2023-09-27:
I tried python=3.8 and open3d=0.17. Can confirm that it works. So when you create the conda environment use python=3.8 and then install open3d=0.17 version