MIT-SPARK/TEASER-plusplus #163: [BUG]

2848385707, 2023-02-28:
**Describe the bug**
A clear and concise description of what the bug is.

**Have you run the unit tests?**

- [x] Yes
- [] No --- then this issue will be closed.

**Installed Dependencies**
Please report the versions of all dependencies you installed, as well as your OS.

**To Reproduce**
Please post the exact steps you executed to reproduce the bug.

**Additional context**
Add any other context about the problem here.
I used my own data set and used scale estimation. I entered my own feature matching pairs for teaser++, and finally the scale estimated by teaser++ was 0 and the rotation matrix was np.ones((1,1,1)). I wanted to know if the feature matching pairs I used were not correct enough. Thank you in advance.
`FPFH generates 3342 putative correspondences.
显示线
Starting scale solver.
Scale estimation complete.
Max core number: 3
Num vertices: 3343
Max Clique of scale estimation inliers: 
Using chain graph for GNC rotation.
Starting rotation solver.
GNC rotation estimation noise bound:inf
GNC rotation estimation noise bound squared:inf
Rotation estimation complete.
Starting translation solver.
Translation estimation complete.
R_teaser: [[1. 0. 0.]
 [0. 1. 0.]
 [0. 0. 1.]]
t_teaser: [-53.17499924   1.02737504   4.07474995]
scale: 0.0
T_teaser: [[  1.           0.           0.         -53.17499924]
 [  0.           1.           0.           1.02737504]
 [  0.           0.           1.           4.07474995]
 [  0.           0.           0.           1.        ]]
`

jingnanshi, 2023-05-22:
It seems like the rotation noise bound set is infinite, which should not occur. Did you explicitly set the parameters for the solver when you use the Python API?

LimHyungTae, 2024-05-02:
If there are no further questions, we will close the issue. Thx!

closed 2024-05-02 by jingnanshi