MIT-SPARK/TEASER-plusplus #164: Installation fails with cmake ..

nofreewill42, 2023-05-05:
I try to install the repository with:

```
git clone --depth 1 https://github.com/MIT-SPARK/TEASER-plusplus.git
cd TEASER-plusplus
mkdir build; cd build
cmake ..
```

And the following is the output of cmake, what should I try to resolve?

TEASER-plusplus/build$ cmake ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting build type to 'Release' as none was specified.
-- Enable printing of diagnostic messages.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/predev/repos/jonatan/aggregate/MULLS/dependent_libraries/TEASER-plusplus/build/googletest-download
[ 11%] Performing download step (download, verify and extract) for 'googletest'
-- File already exists but no hash specified (use URL_HASH):
  file='/home/predev/repos/jonatan/aggregate/MULLS/dependent_libraries/TEASER-plusplus/build/googletest-download/googletest-prefix/src/release-1.8.1.zip'
Old file will be removed and new file downloaded from URL.
-- Downloading...
   dst='/home/predev/repos/jonatan/aggregate/MULLS/dependent_libraries/TEASER-plusplus/build/googletest-download/googletest-prefix/src/release-1.8.1.zip'
   timeout='none'
-- Using src='https://github.com/google/googletest/archive/release-1.8.1.zip'
-- Retrying...
-- Using src='https://github.com/google/googletest/archive/release-1.8.1.zip'
-- Retry after 5 seconds (attempt #2) ...
-- Using src='https://github.com/google/googletest/archive/release-1.8.1.zip'
-- Retry after 5 seconds (attempt #3) ...
-- Using src='https://github.com/google/googletest/archive/release-1.8.1.zip'
-- Retry after 15 seconds (attempt #4) ...
-- Using src='https://github.com/google/googletest/archive/release-1.8.1.zip'
-- Retry after 60 seconds (attempt #5) ...
-- Using src='https://github.com/google/googletest/archive/release-1.8.1.zip'
CMake Error at googletest-download/googletest-prefix/src/googletest-stamp/download-googletest.cmake:159 (message):
  Each download failed!

    error: downloading 'https://github.com/google/googletest/archive/release-1.8.1.zip' failed
         status_code: 1
         status_string: "Unsupported protocol"
         log:
         --- LOG BEGIN ---
         Protocol "https" not supported or disabled in libcurl

  Closing connection -1

  

         --- LOG END ---
         error: downloading 'https://github.com/google/googletest/archive/release-1.8.1.zip' failed
         status_code: 1
         status_string: "Unsupported protocol"
         log:
         --- LOG BEGIN ---
         Protocol "https" not supported or disabled in libcurl
[... 184 lines cut ...]

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 80d45248 v2.4.3 release
Submodule 'tools/clang' (https://github.com/wjakob/clang-cindex-python3) registered for path 'tools/clang'
Cloning into '/home/predev/repos/jonatan/aggregate/MULLS/dependent_libraries/TEASER-plusplus/build/pybind11-src/tools/clang'...
Submodule path 'tools/clang': checked out '6a00cbc4a9b8e68b71caf7f774b3f9c753ae84d5'
[ 33%] No patch step for 'pmc'
[ 44%] Performing update step for 'pmc'
[ 55%] No configure step for 'pmc'
[ 66%] No build step for 'pmc'
[ 77%] No install step for 'pmc'
[ 88%] No test step for 'pmc'
[100%] Completed 'pmc'
[100%] Built target pmc
-- Found PythonInterp: /home/predev/anaconda3/bin/python3.7 (found version "3.7.3") 
-- Found PythonLibs: /home/predev/anaconda3/lib/libpython3.7m.so
-- pybind11 v2.4.3
-- TEASER++ Python binding will be built.
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- LTO enabled
-- Python Interpreter Version: 3.7
-- Configuring incomplete, errors occurred!
See also "/home/predev/repos/jonatan/aggregate/MULLS/dependent_libraries/TEASER-plusplus/build/CMakeFiles/CMakeOutput.log".

jingnanshi, 2023-05-05:
Can you try to see whether you can access the google test release url using your browser?

nofreewill42, 2023-05-05:
If I paste it in the browser, or even wget downloads it without an issue.

nofreewill42, 2023-05-07:
@jingnanshi so is it possible somehow to continue the installation process with having that file downloaded manually?

jingnanshi, 2023-05-22:
@nofreewill42 Can you try the steps outlined in https://stackoverflow.com/questions/29816529/unsupported-protocol-while-download-tar-gz-package? In particular `sudo apt-get install openssl libssl-dev`

nofreewill42, 2023-06-02:
It said the installed versions are already the latest for these.
But tried to install TEASER again and now it went through somehow!
Thanks @jingnanshi

jingnanshi, 2023-06-02:
No problem @nofreewill42 ! It's possible that they have been updated automatically. Let me know if you have any more questions.

closed 2023-06-02 by jingnanshi

nofreewill42, 2023-06-02:
Thank you! Have a nice weekend ! ^^'