MIT-SPARK/TEASER-plusplus #174: undefined reference to `teaser::SolutionEvaluator::SolutionEvaluator(Eigen::Matrix<double, 3, -1, 0, 3, -1> const&, Eigen::Matrix<double, 3, -1, 0, 3, -1> const&, double)'

junxiaofanchen, 2023-11-17:
/usr/bin/ld: CMakeFiles/all_tests.dir/evaluation-test.cc.o: in function `EvaluationTest_Simple_Test::TestBody()':
evaluation-test.cc:(.text+0x2d0): undefined reference to `teaser::SolutionEvaluator::SolutionEvaluator(Eigen::Matrix<double, 3, -1, 0, 3, -1> const&, Eigen::Matrix<double, 3, -1, 0, 3, -1> const&, double)'
/usr/bin/ld: evaluation-test.cc:(.text+0x371): undefined reference to `teaser::SolutionEvaluator::computeErrorMetric(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [test/teaser/CMakeFiles/all_tests.dir/build.make:297: test/teaser/all_tests] Error 1
make[1]: *** [CMakeFiles/Makefile2:582: test/teaser/CMakeFiles/all_tests.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

I get this error when i use make -j16, do anyone know how to fix it?

isqjsnao, 2024-03-19:
> /usr/bin/ld: CMakeFiles/all_tests.dir/evaluation-test.cc.o: in function `EvaluationTest_Simple_Test::TestBody()': evaluation-test.cc:(.text+0x2d0): undefined reference to [...]

请问您如何解决的  我也是这个问题

LimHyungTae, 2024-05-02:
If there are no further questions, we will close the issue. Thx!

hr2894235132, 2024-07-31:
> /usr/bin/ld: CMakeFiles/all_tests.dir/evaluation-test.cc.o: in function `EvaluationTest_Simple_Test::TestBody()': evaluation-test.cc:(.text+0x2d0): undefined reference to [...]

Have you solved it? I have the same problem:undefined reference to `testing::internal::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'

LimHyungTae, 2024-07-31:
Could you elaborate on your problem more? I haven't experienced that kind of problem. How can we reproduce?