MIT-SPARK/TEASER-plusplus #176: [QUESTION]ModuleNotFoundError: No module named 'teaserpp_python.teaserpp_python'

Hana61, 2023-11-23:
I met the following error when testing the python example

(teaser_test) root@something:~/autodl-tmp/TEASER-plusplus/examples/teaser_python_ply# python teaser_python_ply.py
Traceback (most recent call last):
  File "teaser_python_ply.py", line 5, in <module>
    import teaserpp_python
  File "/root/miniconda3/envs/teaser_test/lib/python3.6/site-packages/teaserpp_python/__init__.py", line 1, in <module>
    from .teaserpp_python import *
ModuleNotFoundError: No module named 'teaserpp_python.teaserpp_python'

after 

(teaser_test) root@something:~/autodl-tmp/TEASER-plusplus/build# cd python && pip install .
Processing /root/autodl-tmp/TEASER-plusplus/build/python
  Requirement already satisfied (use --upgrade to upgrade): teaserpp-python==1.0.0 from file:///root/autodl-tmp/TEASER-plusplus/build/python in /root/miniconda3/envs/teaser_test/lib/python3.6/site-packages

plz tell me how to run the python example

YuexinMu, 2024-03-08:
i also get it, but i don't how to slove it

LimHyungTae, 2024-05-02:
Which version do you use? Please remove `/usr/local/include/teaser` and then re-build the package please. I also confronted that issue, but it was my mistake.