MIT-SPARK/TEASER-plusplus #182: [BUG] Can't work with large point clouds (bad_alloc)

Lurvelly, 2024-03-28:
**Describe the bug**
Hi all,
    I'm trying to registry a point clouds with 161817 vertices, but Eigen throw std bad alloc error.
    I found the function "teaser::RobustRegistrationSolver::computeTIMs" trying to alloc a lagre Eigen::Matrix with 3 rows and (N * (N-1) / 2 ) cols. ( In my case, it's 13092289836 cols)
    https://github.com/MIT-SPARK/TEASER-plusplus/blob/baf69d948d77e8fe496a82e2fa3b1f41a9b1156f/teaser/src/registration.cc#L485

**Have you run the unit tests?**
Yes

**Installed Dependencies**

Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.4 LTS
Release:	20.04
Codename:	focal

Eigen: 3.3.7-2
Boost: 1.71.0.0ubuntu2

**To Reproduce**
 
 Just trying to registry a large point clouds.

**Additional context**
[image]

jingnanshi, 2024-04-03:
Please first find correspondences. TEASER++ takes corresponding keypoints instead of raw point clouds.

closed 2024-04-03 by jingnanshi

Lurvelly, 2024-04-03:
> Please first find correspondences. TEASER++ takes corresponding keypoints instead of raw point clouds.

Hi, Jingnan

Thanks for you response. 

Is there any methods that you recommend to extract keypoints from the raw point cloud ? I would like to use TEASER++ find the correct rigid transformation for relocation. But I don't know which method to extract keypoints can work well with TEASER++.