MIT-SPARK/TEASER-plusplus #183: [QUESTION] ament_cmake & ROS2

InguChoi, 2024-04-26:
**Have you read the documentation?**

- [ ] Yes
- [x] No - then this issue will be closed.

Post your theoretical questions / usage questions here.
It it possible to build this pacakge with ament_cmake / ROS2?

LimHyungTae, 2024-05-02:
Normal cmake project 
→ As the warning says, I guess you omit to set Params. Please check it. Here's an example as well:
```
inline void getParams(const double noise_bound, const std::string reg_type, const bool use_robin,
                      teaser::RobustRegistrationSolver::Params &params) {
  params.noise_bound             = noise_bound;
  params.cbar2                   = 1;
  params.estimate_scaling        = false;
  params.use_max_clique          = !use_robin;
  params.rotation_max_iterations = 100;
  params.rotation_gnc_factor     = 1.4;
  if (reg_type == "Quatro") {
    params.rotation_estimation_algorithm =
      teaser::RobustRegistrationSolver::ROTATION_ESTIMATION_ALGORITHM::QUATRO;
    params.inlier_selection_mode         =
      teaser::RobustRegistrationSolver::INLIER_SELECTION_MODE::PMC_EXACT; // HL: PMC_EXACT showed better performance
  } else if (reg_type == "TEASER") {
    params.rotation_estimation_algorithm =
      teaser::RobustRegistrationSolver::ROTATION_ESTIMATION_ALGORITHM::GNC_TLS;
    params.inlier_selection_mode = teaser::RobustRegistrationSolver::INLIER_SELECTION_MODE::PMC_EXACT;
  } else {
    throw std::invalid_argument("Not implemented!");
  }
  params.rotation_cost_threshold = 0.0001;
}
```
And the second one is weird. Did you do 'sudo make install` in your desktop? For your information, I share my `CMakelists.txt` in my ROS2 package:

```
cmake_minimum_required(VERSION 3.10)
project(spark-nss24)

set(CMAKE_CXX_STANDARD 20)

if (NOT CMAKE_BUILD_TYPE)
    set(CMAKE_BUILD_TYPE "Release")
endif ()

find_package(Eigen3 REQUIRED)
find_package(teaserpp REQUIRED)
#find_package(robin REQUIRED)
find_package(PCL REQUIRED)
find_package(pcl_conversions REQUIRED)
find_package(OpenMP REQUIRED)
find_package(small_gicp REQUIRED)

# For ROS2
find_package(ament_cmake REQUIRED)
find_package(nav_msgs REQUIRED)
find_package(rcutils REQUIRED)
find_package(rclcpp REQUIRED)
find_package(rclcpp_components REQUIRED)
find_package(geometry_msgs REQUIRED)
find_package(std_msgs REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(visualization_msgs REQUIRED)
find_package(tf2_ros REQUIRED)
find_package(GTSAM REQUIRED)

include(cmake/CompilerOptions.cmake)

#add_subdirectory(plane_seg_core)
add_subdirectory(robin_core)

include_directories(
    include
    src
    ${PCL_INCLUDE_DIRS}
    /usr/include/rapidjson # RapidJSON's include path
#    plane_seg_core/src
    robin_core/include
)

##########################################################
# Declare a C++ executable
add_executable(inverse_tsfm src/inverse_tsfm.cpp include/area_manager.cpp)
ament_target_dependencies(inverse_tsfm
    rclcpp
    sensor_msgs
    geometry_msgs
    std_msgs
    visualization_msgs
    Eigen3
    teaserpp
    PCL
    pcl_conversions
    OpenMP
)
target_link_libraries(inverse_tsfm
    Eigen3::Eigen
    Eigen3::Eigen
    teaserpp::teaser_registration
    teaserpp::teaser_features
    teaserpp::teaser_io
    ${PCL_LIBRARIES}
    OpenMP::OpenMP_CXX
#    plane_seg::plane_seg_core
    robin::robin_core
    xenium
    pmc
    gtsam
    small_gicp
)

add_executable(main src/main.cpp src/nss_reg_server.cpp src/KISSMatcher.cpp include/area_manager.cpp)
ament_target_dependencies(main
    rclcpp
    sensor_msgs
    geometry_msgs
    std_msgs
    visualization_msgs
    Eigen3
    teaserpp
    PCL
    pcl_conversions
    OpenMP
)

target_link_libraries(main
    Eigen3::Eigen
    teaserpp::teaser_registration
    teaserpp::teaser_features
    teaserpp::teaser_io
    ${PCL_LIBRARIES}
    OpenMP::OpenMP_CXX
#    plane_seg::plane_seg_core
    robin::robin_core
    xenium
    pmc
    gtsam
    small_gicp
)

install(TARGETS
#    json_load_test
    inverse_tsfm
#    ransac_preprocessing
    main
    DESTINATION lib/${PROJECT_NAME})
install(DIRECTORY launch rviz DESTINATION share/${PROJECT_NAME}/)

ament_package()
```

InguChoi, 2024-05-09:
Thank you for giving an answer Dr. Lim!

Original CMakeLists.txt`s contents is below.

```
cmake_minimum_required(VERSION 3.10)
project(wm_teaser_plusplus)

set(CMAKE_CXX_STANDARD 14)

set(CMAKE_MODULE_PATH "${CMAKE_CURRENT_SOURCE_DIR}/cmake/modules/" ${CMAKE_MODULE_PATH})
set(CMAKE_LIBRARY_PATH ${CMAKE_LIBRARY_PATH} "/home/wm_server/TEASER-plusplus/build/teaser")

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
# uncomment the following section in order to fill in
# further dependencies manually.
# find_package(<dependency> REQUIRED)

find_package(rclcpp REQUIRED)
# find_package(eigen3_cmake_module REQUIRED)
find_package(Eigen3 REQUIRED)
find_package(teaserpp REQUIRED)
find_package(PCL 1.8 REQUIRED)
# find_package(tinyply)

# Find package in custom directory
# find_package(teaserpp REQUIRED PATHS /home/wm_server/TEASER-plusplus)

# Check if package was found
if(teaserpp_FOUND)
    message(STATUS "Found teaserpp package")
else()
    message(FATAL_ERROR "teaserpp package not found")
endif()

# include_directories(${teaserpp_INCLUDE_DIRS})
ament_export_include_directories(${teaserpp_INCLUDE_DIRS})

find_package(OpenMP)
if (OPENMP_FOUND)
    set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${OpenMP_C_FLAGS}")
    set (CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${OpenMP_CXX_FLAGS}")
endif()

include_directories(${PCL_INCLUDE_DIRS})
link_directories(${PCL_LIBRARY_DIRS})
add_definitions(${PCL_DEFINITIONS})

include_directories(
        include
        src
)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # comment the line when a copyright and license is added to all source files
  set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # comment the line when this package is in a git repo and when
  # a copyright and license is added to all source files
  set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

add_executable(wm_teaser_plusplus src/wm_teaser_plusplus.cpp)
ament_target_dependencies(wm_teaser_plusplus rclcpp teaserpp Eigen3 OpenMP)
# ament_target_dependencies(wm_teaser_plusplus rclcpp teaserpp::teaser_registration teaserpp::teaser_io Eigen3 OpenMP)

install(TARGETS
  wm_teaser_plusplus
  DESTINATION lib/${PROJECT_NAME})

install(
  DIRECTORY include/ dataset/
  DESTINATION include
)

# ament_export_dependencies(eigen3_cmake_module)
# ament_export_dependencies(Eigen3)

ament_package()

```

---

After changing the CMakeLists.txt`s contents in reference to your advice,
```
cmake_minimum_required(VERSION 3.10)
project(wm_teaser_plusplus)

set(CMAKE_CXX_STANDARD 20)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(Eigen3 REQUIRED)
find_package(teaserpp REQUIRED)
find_package(PCL REQUIRED)
find_package(pcl_conversions REQUIRED)
find_package(GTSAM REQUIRED)
find_package(OpenMP)

include_directories(${PCL_INCLUDE_DIRS} include)
link_directories(${PCL_LIBRARY_DIRS})
add_definitions(${PCL_DEFINITIONS})

add_executable(wm_teaser_plusplus src/wm_teaser_plusplus.cpp)
ament_target_dependencies(wm_teaser_plusplus
  rclcpp
  teaserpp
  Eigen3
  PCL
  pcl_conversions
  OpenMP
)

target_include_directories(wm_teaser_plusplus PUBLIC
  $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>
  $<INSTALL_INTERFACE:include>
)

target_link_libraries(wm_teaser_plusplus
  Eigen3::Eigen
  ${PCL_LIBRARIES}
  teaserpp::teaser_registration
  # teaserpp::teaser_features
  teaserpp::teaser_io
  OpenMP::OpenMP_CXX
  #    plane_seg::plane_seg_core
  # robin::robin_core
  # xenium
  pmc
  gtsam
  # small_gicp
)

install(TARGETS
  wm_teaser_plusplus
  DESTINATION lib/${PROJECT_NAME})

  install(
    DIRECTORY "include/"
    DESTINATION include
  )

  if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # comment the line when a copyright and license is added to all source files
  set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # comment the line when this package is in a git repo and when
  # a copyright and license is added to all source files
  set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
```

I can build it!!!

[image]

Thank you so much!!!

InguChoi, 2024-05-09:
If I could share my package, would it help people who want to build teaser++ using ros2? If it helps, I can share the base package that builds with ament_cmake.

LimHyungTae, 2024-05-09:
Thanks, but I think we maintain TEASER++ as a stand-alone library (what if ROS3 comes? Then, should we modify the codes again?), so it's fine :+1:

jingnanshi, 2024-05-09:
thanks @LimHyungTae for helping out and thanks @InguChoi for your interest. Closing this now.

closed 2024-05-09 by jingnanshi