MIT-SPARK/TEASER-plusplus #186: [QUESTION] when  use vscode debug,I get some errors .

Runasia, 2024-05-20:
**Have you read the documentation?**

- [🐕] Yes
- [ ] No - then this issue will be closed.

Post your theoretical questions / usage questions here.

I am a beginner,, and thank you for your assistance.

I want to debug code ,but 
[image]
this is my cmakelist 
cmake_minimum_required(VERSION 3.10)
project(teaser_cpp_cptry)

set(CMAKE_CXX_STANDARD 14)

FIND_PACKAGE( OpenMP REQUIRED)
find_package(Eigen3 REQUIRED)
find_package(teaserpp REQUIRED)

# Change this line to include your own executable file
add_executable(teaser_cpp_cptry teaser_cpp_cptry.cc)

# Link to teaserpp & Eigen3
target_link_libraries(teaser_cpp_cptry Eigen3::Eigen teaserpp::teaser_registration teaserpp::teaser_io OpenMP::OpenMP_CXX)

# Copy the data files to build directory
file(COPY ${CMAKE_SOURCE_DIR}/../example_data/
        DESTINATION ${CMAKE_BINARY_DIR}/example_data/
        FILES_MATCHING
        PATTERN *.ply)

I can use command --cmake .. && make     can  get  teaser_cpp_cptry 
[image]

so i need some change or.....  thanks!
I will continue to search for solutions or methods to configure .

jingnanshi, 2024-05-20:
just to clarify your CMakeList file works in terminal but not in VSCode?

Runasia, 2024-05-21:
> just to clarify your CMakeList file works in terminal but not in VSCode?

Yes! but i'm not sure is CMakeList Mistake or something .

Runasia, 2024-05-22:
I find  use  "-fopenmp" in task.json can solve openMP ，but  another undefined still exists.

[image]

Runasia, 2024-05-27:
It's gcc or g++ problem?..

Runasia, 2024-05-27:
Can't do anything ,  how to  Configure debug environment in vscode ... 
[image]
due to this ..

Runasia, 2024-05-28:
> the std::bad_alloc might be due to memory overflow. Regarding VSCode, there seems to be a linker error but unfortunately I am not familiar with VSCode configuration.

good night ! thinks for your reply , can i know how idea or something to debug code ?  I

Runasia, 2024-05-28:
> the std::bad_alloc might be due to memory overflow. Regarding VSCode, there seems to be a linker error but unfortunately I am not familiar with VSCode configuration.

That 's maybe due to  that i run code in VMware and memory is too small...

closed 2024-08-05 by Runasia