MIT-SPARK/TEASER-plusplus #189: [Has anyone ever use teaser to marge more than 2 pointcloud together?]

Monsurat-Onabajo, 2024-06-26:
**Have you read the documentation?**

- [x] Yes
- [] No - then this issue will be closed.

Post your theoretical questions / usage questions here.
I'm attempting to merge six point clouds using Teaser ++. Initially, I successfully stitched two of them together. My plan was to stitch pairs to create three merged point clouds, and then continue merging these results step-by-step until all were combined. Unfortunately, this method isn't producing the expected outcomes. Has anyone tried a similar approach with multiple point clouds? If so, how did it work out for you?

LimHyungTae, 2024-08-25:
Could you provide some snapshots or something for the unexpected outcomes please?

Monsurat-Onabajo, 2024-08-25:
Hello, I was able to get it to work finally, I used a for loop and merge two at a time

jingnanshi, 2024-09-24:
@Monsurat-Onabajo Great to hear!

closed 2024-09-24 by jingnanshi