MIT-SPARK/TEASER-plusplus #193: Minimal C++ example，Compilation error

BoBoINVICTUS, 2024-09-06:
My system is ubuntu 22.04
When I try run “teaser_cpp_ply.cc”，a compilation error has occurred:
:~/TEASER-plusplus/examples/teaser_cpp_ply/build$ cmake .. && make
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Configuring done
CMake Error in CMakeLists.txt:
  Imported target "teaserpp::teaser_registration" includes non-existent path

    "/home/xiaowang/TEASER-plusplus/build/teaser/CMakeFiles/Export/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

BoBoINVICTUS, 2024-09-06:
I solved the problem, but why is it that my output is not the same as the one you provided?
[image]
[image]

jingnanshi, 2024-09-16:
Please do not use GPT responses in your issues. Regarding the difference in estimated rotations and translations, note that the example generates noises and outliers randomly.

closed 2024-09-16 by jingnanshi