MIT-SPARK/TEASER-plusplus #206: [QUESTION]段错误

lalawinter, 2025-07-23:
./robot_test_ply 
	Read 74712 total vertices 
	Read 74712 total vertices 

----------------------enter solver
src.size() : 224136
target.size() : 224136
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
请问算法是无法支撑这个量级的计算吗

XiaXingLuo, 2025-09-15:
I also meet this problem, have you solved it?

ryancwicks-voyis, 2025-09-24:
I have a similiar problem. The program is crashing because Eigen is trying to allocate more memory than the system has during a resize.

The root of the problem is the TIM's: 

 /**
   * Given a 3-by-N matrix representing points, return Translation Invariant Measurements (TIMs)
   * @param v a 3-by-N matrix
   * @return a 3-by-(N-1)*N matrix representing TIMs
   */
  Eigen::Matrix<double, 3, Eigen::Dynamic>
  computeTIMs(const Eigen::Matrix<double, 3, Eigen::Dynamic>& v,
              Eigen::Matrix<int, 2, Eigen::Dynamic>* map);

For your matrix of 74712 points, this tries to allocate 62 GB of memory. I'm not a developer on this project, so I don't know if there is a way around this problem short of decimating your data very heavily.